coolsnowwolf/lede #1358: R7800 L2TP拨号四个小时断一次，同时会处于死机状态

panyaolin, 2019-05-28:
@coolsnowwolf 
系统日志在下面
[my.log](https://github.com/coolsnowwolf/lede/files/3227512/my.log)

[my2.log](https://github.com/coolsnowwolf/lede/files/3227544/my2.log)

四个小时左右会down一次，表现为L2TP拨号先挂掉，这时候还能访问内网的设备，也能访问路由器后台，但是只能看到eth0.2在工作，过一会就连路由器后台都进不去，内网设备也互相ping不通，只能通过重启路由器才能恢复。

lan542662, 2019-06-12:
为我家7800打call

hzdrro, 2019-07-08:
lz解决了吗？ @panyaolin 
同zju，mt7621at也有这个问题。但单核的mt7620和qca9558都能正常工作。另外，mt7621at在pandorabox和padavan下却没问题。

panyaolin, 2019-07-08:
> lz解决了吗？ @panyaolin
> 同zju，mt7621at也有这个问题。但单核的mt7620和qca9558都能正常工作。另外，mt7621at在pandorabox和padavan下却没问题。

校友可还行...后面没解决，但是我测试了，换3.x内核版本的openwrt就是好的，不过也是间歇性丢包（和官方固件一样，官方固件也是3.x版本的openwrt魔改的），根据log上面看，我觉得应该是内核的某些功能（比如daemon进程）和xl2tp的冲突了，像是有个daemon进程会在连上几个小时后，把xl2tp的tunnel给kill掉，然后xl2tp的守护进程就起来了，反复dial out，拨不出去就一直在打log，反复重拨就直接把整个内核都给拖炸了。
你7621刷padavan用着得了，目前看来Openwrt的4.x内核就是有问题，我跑去国外的Openwrt论坛发帖贴log了，也没人理我，都不承认内核的问题，扯着说是我是从lean的仓库编译的，不是他们官方的，肯定是第三方代码的问题。关键我中间也用官方仓库的源码编译过，测试过，全都是一样的，几个小时就炸。我寻思着lean这个仓库，怎么改也就加了点应用层面的东西而已，这肯定是官方的内核哪里没做好。
之前一台R6800（也是7621的）用padavan拨号都很稳，以及一个newifi mini，7620的，padavan一样稳定拨号，后面都送人了。7800原厂固件在玉泉基本属于没法用的情况，两小时改一次MTU，不然就严重丢包。padavan也没做这个型号的适配，pandorabox没集成ssr。后面我直接把7800卖掉了，目前换了MikroTik的RB4011，拨号和静态路由都很稳，就是自带的无线略垃圾，160MHz的频段支持和shit一样。
个人建议在我们校网下，设备不多7620的路由器凑合用用就行了，就是百兆口有点难受，设备多7621刷个padavan也挺好使，想稳愿意学新东西就上mikrotik的普通的路由器（不带无线，千兆入门差不多的那种，比如RB750GR3），再挂个性能好一点的AP，这个搭配在校网下面稳定性真的锤爆网件华硕（顺带提一嘴我之前用过K3刷梅林，自带的L2TP拨号也得改MTU，不然上不了网；而且玉泉的网最近是越来越垃圾了，对阿里的dns长ping，延迟是很低，2ms，但是五个包丢三个你敢信，真的看不懂在搞什么）。

hzdrro, 2019-07-09:
> 校友可还行...后面没解决，但是我测试了，换3.x内核版本的openwrt就是好的，不过也是间歇性丢包（和官方固件一样，官方固件也是3.x版本的openwrt魔改的），根据log上面看，我觉得应该是内核的某些功能（比如daemon进程）和xl2tp的冲突了，像是有个daemon进程会在连上几个小时后，把xl2tp的tunnel给kill掉，然后xl2tp的守护进程就起来了，反复dial [...]

感谢大佬回复。
我组了台软路由，同样是4.14内核的openwrt，已经稳定运行18h了，Ping百度丢包0.1%~0.2%。看来跟cpu架构也有关系，x86没问题，arm和mips跟openwrt\xl2tpd有冲突？已经完全搞不懂了

panyaolin, 2019-07-09:
> 感谢大佬回复。
> 我组了台软路由，同样是4.14内核的openwrt，已经稳定运行18h了，Ping百度丢包0.1%~0.2%。看来跟cpu架构也有关系，x86没问题，arm和mips跟openwrt\xl2tpd有冲突？已经完全搞不懂了

软路由功耗还是略大了些，而且ether口多的板子不便宜，少的板子需要自己接pcie扩展卡，之前倒是想组一个，但是看了看体积功耗就溜了（当然有新平台功耗更低的，就是价格不太能接受）。另外x86的openwrt镜像最好自己编译，不要用koolshare的。

closed 2019-10-18 by coolsnowwolf