coolsnowwolf/lede #1363: uci_load_validate :not fond 错误

ope087, 2019-05-30:
早几个月openwrt主线里的package/system/procd/files/procd.sh,里添加了uci_load_validate函数,但我们这个版本里并没有合并这个版本,而且新更新的nft-qos又有这个函数,导致加载配置的时候报错且功能不正常.我只在procd.sh里添加了uci_load_validate这段,测试了nft-qos是可以正常运行的.

coolsnowwolf, 2019-05-30:
可以帮忙提交 pull 么，谢谢

zxlhhyccc, 2019-05-31:
> 可以帮忙提交 pull 么，谢谢

大佬，我pull给你，他加的应该是这段代码:
uci_load_validate() {
	local _package="$1"
	local _type="$2"
	local _name="$3"
	local _function="$4"
	local _option
	local _result
	shift; shift; shift; shift
	for _option in "$@"; do
		eval "local ${_option%%:*}"
	done
	uci_validate_section "$_package" "$_type" "$_name" "$@"
	_result=$?
	[ -n "$_function" ] || return $_result
	eval "$_function \"\$_name\" \"\$_result\""
}
[image]

zxlhhyccc, 2019-05-31:
[procd.zip](https://github.com/coolsnowwolf/lede/files/3241299/procd.zip)

closed 2019-05-31 by coolsnowwolf