coolsnowwolf/lede #1364: 求教rdp远程桌面端口转发失效问题

WROIATE, 2019-05-30:
[image]

[image]

路由器sbr ac1750
如图，这是直接编译安装完openwrt以后什么都不改直接设置端口转发，结果是被远程的电脑主机是有反应的，远程端这边也没显示失败，但是就是大部分情况下黑屏，偶尔会出现一些图像或者主题色覆盖，在黑屏情况下瞎点后，自己去看被远程主机，发现主机是会被影响，比如我瞎点开了chrome。不太清楚是什么情况，在pandora上相同配置完全没问题，希望有大佬能帮忙解答一下

WROIATE, 2019-07-15:
求大佬解答==我现在是真的哭了用的远程是微软自带的mstsc

alexyw, 2019-08-01:
我也出现了类似情况！不光黑屏，等半天进入后还非常慢，比官方openwrt效果差多了。
昨晚我重新编译了一次，去掉adbdy、upnpd，勾了ssr，再次设置，这次端口转发就完全失效了。查看iptables，跟之前没区别，不晓得哪个环节出错了……求助

WROIATE, 2019-09-02:
大佬们都不知道吗。。。哭哭

closed 2019-10-18 by coolsnowwolf