coolsnowwolf/lede #1372: 编译十二遍没成功，贴个issues出来大家帮忙看看。

236212005, 2019-06-02:
**环境：**
1.系统为ubuntu 18.04；
2.readme.md里说的依赖全装了；
3.有梯子有proxychains4；
4.试过make download命令；
5.-j1，-j4，-j8都不行；
6.以为三自己选择的组件有问题，rm -f .config，采用menuconfig的默认配置依然报错；

**错误日志如下：**

> make[5]: Leaving directory '/home/shirui/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.178'
touch /home/shirui/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.178/.modules
make -C image compile TARGET_BUILD=
make[5]: Entering directory '/home/shirui/lede/target/linux/x86/image'
make[5]: Nothing to be done for 'compile'.
make[5]: Leaving directory '/home/shirui/lede/target/linux/x86/image'
make[4]: Leaving directory '/home/shirui/lede/target/linux/x86'
make[3]: Leaving directory '/home/shirui/lede/target/linux'
time: target/linux/compile#458.31#56.56#499.71
make[2]: Leaving directory '/home/shirui/lede'
export MAKEFLAGS= ;make -w -r diffconfig
make[2]: Entering directory '/home/shirui/lede'
make[2]: Leaving directory '/home/shirui/lede'
make[2]: Entering directory '/home/shirui/lede'
rm -rf /home/shirui/lede/staging_dir/target-x86_64_musl/root-x86
make[2]: Leaving directory '/home/shirui/lede'
make[2]: Entering directory '/home/shirui/lede'
make[3]: Entering directory '/home/shirui/lede/package/libs/libjson-c'
make[3]: Leaving directory '/home/shirui/lede/package/libs/libjson-c'
time: package/libs/libjson-c/host-compile#0.23#0.11#0.30
make[3]: Entering directory '/home/shirui/lede/package/libs/libubox'
make[3]: Leaving directory '/home/shirui/lede/package/libs/libubox'
time: package/libs/libubox/host-compile#0.75#0.48#1.18
make[3]: Entering directory '/home/shirui/lede/package/system/opkg'
CFLAGS="-O2 -I/home/shirui/lede/staging_dir/host/include -I/home/shirui/lede/staging_dir/hostpkg/include -I/home/shirui/lede/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/shirui/lede/staging_dir/host/include -I/home/shirui/lede/staging_dir/hostpkg/include -I/home/shirui/lede/staging_dir/target-x86_64_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/shirui/lede/staging_dir/host/lib -L/home/shirui/lede/staging_dir/hostpkg/lib -L/home/shirui/lede/staging_dir/target-x86_64_musl/host/lib" make  -C /home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f  
make[4]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[5]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[6]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
libbb/CMakeFiles/bb.dir/flags.make:9: *** target pattern contains no '%'.  Stop.
make[6]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
CMakeFiles/Makefile2:90: recipe for target 'libbb/CMakeFiles/bb.dir/all' failed
make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2
make[5]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
Makefile:103: recipe for target '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/.built' failed
make[3]: *** [/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/.built] Error 2
make[3]: Leaving directory '/home/shirui/lede/package/system/opkg'
time: package/system/opkg/host-compile#0.60#0.30#0.87
package/Makefile:107: recipe for target 'package/system/opkg/host/compile' failed
make[2]: *** [package/system/opkg/host/compile] Error 2
make[2]: Leaving directory '/home/shirui/lede'
package/Makefile:103: recipe for target '/home/shirui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/shirui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shirui/lede'
/home/shirui/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

signigelchan, 2019-06-02:
try below step:
```
1. ./scripts/feeds update -a

2. ./scripts/feeds install -a

3. make V=s clean

4. make menuconfig

5. make V=s clean

6. make -j2 V=s download

7. make -j2 V=s
```

zcluo, 2019-06-02:
建议下下来代码先别做配置直接编一遍，环境问题先解决再make menuconfig
发自我的 iPhone

SasamiMasakiJurai, 2019-06-02:
make[6]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
CMakeFiles/Makefile2:90: recipe for target 'libbb/CMakeFiles/bb.dir/all' failed
make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2

bb.dir出错

236212005, 2019-06-02:
> try below step:
> 
> ```
> 1. ./scripts/feeds update -a
> 
> 2. ./scripts/feeds install -a
> 
> 3. make V=s clean
> 
> 4. make menuconfig
> 
> 5. make V=s clean
> 
> 6. make -j2 V=s download
> 
> 7. make -j2 V=s
> ```

I got this when I followed your steps:

>make[4]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[5]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
make[6]: Entering directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
libbb/CMakeFiles/bb.dir/flags.make:9: *** target pattern contains no '%'.  Stop.
make[6]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
CMakeFiles/Makefile2:90: recipe for target 'libbb/CMakeFiles/bb.dir/all' failed
make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2
make[5]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
Makefile:129: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
Makefile:103: recipe for target '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/.built' failed
make[3]: *** [/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f/.built] Error 2
make[3]: Leaving directory '/home/shirui/lede/package/system/opkg'
time: package/system/opkg/host-compile#0.60#0.34#0.91
package/Makefile:107: recipe for target 'package/system/opkg/host/compile' failed
make[2]: *** [package/system/opkg/host/compile] Error 2
make[2]: *** Waiting for unfinished jobs....
...........................
make[3]: Leaving directory '/home/shirui/lede/package/libs/elfutils'
time: package/libs/elfutils/compile#93.00#17.12#104.44
make[2]: Leaving directory '/home/shirui/lede'
package/Makefile:103: recipe for target '/home/shirui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/shirui/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shirui/lede'
Build failed - please re-run with -j1 to see the real error message
/home/shirui/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 1

236212005, 2019-06-02:
> 建议下下来代码先别做配置直接编一遍，环境问题先解决再make menuconfig 发自我的 iPhone
> […](#)

谢谢你的建议。目前看来没什么效果，我明天再换台机器是一下，情况应该会更明朗。

236212005, 2019-06-02:
> make[6]: Leaving directory '/home/shirui/lede/build_dir/hostpkg/opkg-2017-12-07-3b417b9f'
> CMakeFiles/Makefile2:90: recipe for target 'libbb/CMakeFiles/bb.dir/all' failed
> make[5]: *** [libbb/CMakeFiles/bb.dir/all] Error 2
> 
> bb.dir出错

我也是这么认为的，但是经过搜索以后我没有发现这个文件，所以我认为是源代码某处写到了这个东西，我试着去找过，但是代码量太过庞大，所以我无奈只能来这里发起issues，希望能够得到大家的建议，如果能够告诉我源码哪里有问题，我至少可以先在本地将其修改掉。或者有人知道这是环境哪里的问题。

fuqiang03, 2019-06-03:
Ubuntu 18.04 的编译环境和其他版本的不一样，需要多添加几个运行库。百度上应该有资料
[image]
[image]

hanxi, 2019-06-03:
你也够厉害的,编译了12遍,还不如装个docker,用Ubuntu14.04来编译.

236212005, 2019-06-04:
> Ubuntu 18.04 的编译环境和其他版本的不一样，需要多添加几个运行库。百度上应该有资料
> Ubuntu 18.04是肯定能编译的 你的错误应该是依赖库不全
> [image]
> [image]
> [image]
> [image]

你是正确的。之前虽然有怀疑，但是未确定。昨天用虚拟机装了14.04LTS来编译，错误是不同的。主要体现在有些依赖包下载失败，导致编译失败。因为昨天我把梯子也换成了新加坡节点。今天换成u14.04lts+日本梯子试下。现在还不确定依赖包的404是梯子节点问题还是服务不可用，慢慢试吧。

这个问题我来慢慢定位吧。所有问题必定都有个真相嘛。。。issue我就先close掉吧。希望这些留言能对后来者有一定启发。再次谢谢热心的各位。

closed 2019-06-04 by 236212005

236212005, 2019-06-04:
结论：
1.要使用u14系统；
2.梯子可能会由于所在区域不同，导致某些依赖无法下载；

本issue问题解决：
ubuntu14.10+日本梯子

再次谢谢大家的帮助。上图供后人借鉴：

[image]

hcym, 2019-06-04:
没有的事，一直1804，无论大雕的官方的还有老毛子都正常编译，梯子倒是重中之重

fuqiang03, 2019-06-04:
我都没有梯子  家里穷啊