coolsnowwolf/lede #1373: 基于最新源码编译出的固件55r不能运行，大家有遇到么？

signigelchan, 2019-06-02:
- 2915c44节点编译出来的没有问题，从b5daaad以上节点还原55r-plus的软件包之后，编译出来的固件都不能正常运行
- ssh之后发现默认木有给执行权限，chmod +x 之后，使用命令/etc/init.d/shadowsockr start ，iptable提示 iptables v1.8.2 (legacy): Set china doesn't exist.

附截图：
[image]

signigelchan, 2019-06-03:
> #### 问题已解决，目录及文件的默认权限问题。

closed 2019-06-03 by signigelchan

azazjjyjjy, 2019-08-29:
> > #### 问题已解决，目录及文件的默认权限问题，感谢各位相助！。

iptables v1.8.2 (legacy): Set china doesn't exist. 这个后来怎么解决的？ 我看了下我的目录权限正常的

signigelchan, 2019-08-29:
> > > #### 问题已解决，目录及文件的默认权限问题，感谢各位相助！。
> 
> iptables v1.8.2 (legacy): Set china doesn't exist. 这个后来怎么解决的？ 我看了下我的目录权限正常的

lede世界和平期间，尝试恢复删除的55r包，从windows复制文件到linux，会丢失文件&目录权限。源码编译前，保证文件&目录有+x权限就好了。