coolsnowwolf/lede #1379: 新版本更新问题求助

lxwq, 2019-06-04:
怎麼才能将已经克隆到本地的代码更新到新版，每次都重新克隆编译好慢啊，

adamw92, 2019-06-04:
以下转自小白mj写给麻瓜看的复读机编译教程 v1.8

> 以后二次编译：
cd lede                                                          进入LEDE目录
git pull                                                         同步更新大雕源码
./scripts/feeds update -a && ./scripts/feeds install -a          更新FEEDS
rm -rf ./tmp && rm -rf .config                                   清除编译配置和缓存
make menuconfig                                                  进入编译配置菜单
make -jn V=s                                                     n=线程数+1，例如4线程的I5填-j5，开始编译

>

qyh214, 2019-06-05:
`rm -rf ./tmp && rm -rf .config 清除编译配置和缓存`并不需要每次都做的，大雕的readme里有写的

lxwq, 2019-06-11:
多谢各位，

lxwq, 2019-06-27:
多谢各位，

closed 2019-06-27 by lxwq