coolsnowwolf/lede #1387: 今天git pull后编译错误

hgh188, 2019-06-11:
c1: some warnings being treated as errors
make[9]: *** [scripts/Makefile.build:327: drivers/net/ethernet/mediatek/mtk_eth_soc.o] Error 1
make[8]: *** [scripts/Makefile.build:585: drivers/net/ethernet/mediatek] Error 2
make[7]: *** [scripts/Makefile.build:585: drivers/net/ethernet] Error 2
make[6]: *** [scripts/Makefile.build:585: drivers/net] Error 2
make[5]: *** [Makefile:1038: drivers] Error 2
make[5]: Leaving directory '/home/a/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.123'
make[4]: *** [Makefile:27: /home/a/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.123/.image] Error 2
make[4]: Leaving directory '/home/a/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/a/lede/target/linux'
time: target/linux/install#3.83#2.09#6.89
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/a/lede'
make[1]: *** [target/Makefile:19: /home/a/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/a/lede'
make: *** [/home/a/lede/include/toplevel.mk:218：world] 错误 2

hgh188, 2019-06-11:
可能是我自己空间不够

KevinMX, 2019-06-11:
遇到了同样问题：
```drivers/net/ethernet/mediatek/mtk_eth_soc.c: In function 'fe_do_ioctl':
drivers/net/ethernet/mediatek/mtk_eth_soc.c:1416:10: error: implicit declaration of function 'phy_ethtool_ioctl'; did you mean 'phy_ethtool_sset'? [-Werror=implicit-function-declaration]
   return phy_ethtool_ioctl(priv->phy_dev,
          ^~~~~~~~~~~~~~~~~
          phy_ethtool_sset
cc1: some warnings being treated as errors
scripts/Makefile.build:326: recipe for target 'drivers/net/ethernet/mediatek/mtk_eth_soc.o' failed
make[9]: *** [drivers/net/ethernet/mediatek/mtk_eth_soc.o] Error 1
scripts/Makefile.build:585: recipe for target 'drivers/net/ethernet/mediatek' failed
make[8]: *** [drivers/net/ethernet/mediatek] Error 2
scripts/Makefile.build:585: recipe for target 'drivers/net/ethernet' failed
make[7]: *** [drivers/net/ethernet] Error 2
scripts/Makefile.build:585: recipe for target 'drivers/net' failed
make[6]: *** [drivers/net] Error 2
Makefile:1038: recipe for target 'drivers' failed
make[5]: *** [drivers] Error 2
make[5]: Leaving directory '/home/mx/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.123'
Makefile:26: recipe for target '/home/mx/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.123/.image' failed
make[4]: *** [/home/mx/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.123/.image] Error 2
make[4]: Leaving directory '/home/mx/lede/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/mx/lede/target/linux'
time: target/linux/install#2.78#0.84#3.44
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/mx/lede'
target/Makefile:19: recipe for target '/home/mx/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/mx/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/mx/lede'
/home/mx/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```
我的是为k2p编译，使用的驱动来自这里：https://github.com/MeIsReallyBa/k2p-openwrt-mt7615_5.0.2.0
是驱动问题？
[build.log](https://github.com/coolsnowwolf/lede/files/3276642/build.log)

KevinMX, 2019-06-12:
今天重新git pull后已经恢复正常。

closed 2019-06-12 by hgh188