coolsnowwolf/lede #1388: X86 的QB 在修复后好像还是不能编译，提示如下

hxlls, 2019-06-12:
checking for Qt5 qmake >= 5.5.1... not found
configure: error: Could not find qmake
Makefile:57: recipe for target '/home/ling/lede/build_dir/target-x86_64_musl/qBittorrent-4.1.3/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/ling/lede/build_dir/target-x86_64_musl/qBittorrent-4.1.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/ling/lede/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#16.65#3.30#262.84
package/Makefile:107: recipe for target 'package/lean/qBittorrent/compile' failed
make[2]: *** [package/lean/qBittorrent/compile] Error 2
make[2]: Leaving directory '/home/ling/lede'
package/Makefile:103: recipe for target '/home/ling/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ling/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ling/lede'
/home/ling/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
ling@ubuntu:~/lede$

zxlhhyccc, 2019-06-12:
空间满了吧

hxlls, 2019-06-12:
ling@ubuntu:~/lede$ df -hl
文件系统        容量  已用  可用 已用% 挂载点
udev            979M     0  979M    0% /dev
tmpfs           200M  3.1M  197M    2% /run
/dev/sda2       124G   23G   95G   20% /
tmpfs          1000M     0 1000M    0% /dev/shm
tmpfs           5.0M     0  5.0M    0% /run/lock
tmpfs          1000M     0 1000M    0% /sys/fs/cgroup
/dev/sda1       511M  3.6M  508M    1% /boot/efi
tmpfs           200M     0  200M    0% /run/user/1000
还有95G空间，怎么会满呢？

zxlhhyccc, 2019-06-12:
> ling@ubuntu:~/lede$ df -hl
> 文件系统 容量 已用 可用 已用% 挂载点
> udev 979M 0 979M 0% /dev
> tmpfs 200M 3.1M 197M 2% /run
> /dev/sda2 124G 23G 95G 20% /
> tmpfs 1000M 0 1000M 0% /dev/shm
> tmpfs 5.0M 0 5.0M 0% /run/lock [...]

我的也报错：
/home/lin/le/k3/staging_dir/target-arm_cortex-a9_musl_eabi/usr/include/openssl/ossl_typ.h:93:16: note: forward declaration of 'EVP_PKEY {aka struct evp_pkey_st}'
 typedef struct evp_pkey_st EVP_PKEY;
                ^~~~~~~~~~~
Makefile:24121: recipe for target '.obj/qsslcertificate_openssl.o' failed
make[7]: *** [.obj/qsslcertificate_openssl.o] Error 1
make[7]: Leaving directory '/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0/qtbase/src/network'
Makefile:239: recipe for target 'sub-network-make_first' failed
make[6]: *** [sub-network-make_first] Error 2
make[6]: Leaving directory '/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0/qtbase/src'
Makefile:46: recipe for target 'sub-src-make_first' failed
make[5]: *** [sub-src-make_first] Error 2
make[5]: Leaving directory '/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0/qtbase'
Makefile:63: recipe for target 'module-qtbase-make_first' failed
make[4]: *** [module-qtbase-make_first] Error 2
make[4]: Leaving directory '/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0'
Makefile:206: recipe for target '/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0/.built' failed
make[3]: *** [/home/lin/le/k3/build_dir/target-arm_cortex-a9_musl_eabi/qt-everywhere-opensource-src-5.8.0/.built] Error 2
make[3]: Leaving directory '/home/lin/le/k3/package/lean/qt5'
time: package/lean/qt5/compile#1323.44#437.89#1995.42
package/Makefile:107: recipe for target 'package/lean/qt5/compile' failed
make[2]: *** [package/lean/qt5/compile] Error 2
make[2]: Leaving directory '/home/lin/le/k3'
package/Makefile:103: recipe for target '/home/lin/le/k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/lin/le/k3/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/le/k3'
/home/lin/le/k3/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

xinxijishuwyq, 2019-06-13:
安装qmake呢

xinxijishuwyq, 2019-06-13:
Debian/Ubuntu尝试
sudo apt update
sudo apt install qt5-default

Arch Linux尝试
sudo pacman -Syu
sudo pacman -S qt5-base

shenzt68, 2019-06-13:
[image]
这个怎么回事？node.js后面没箭头，不能选择node.js

xinxijishuwyq, 2019-06-13:
> [image]
> 这个怎么回事？node.js后面没箭头，不能选择node.js

不支持的你的cpu。。。(

shenzt68, 2019-06-13:
k3不支持吗？

hxlls, 2019-06-13:
QT5 已经安装好测试过，还是不行，貌似提示是少了某个文件。。
configure: error: Could not find qmake
Makefile:57: recipe for target '/home/ling/lede/build_dir/target-x86_64_musl/qBittorrent-4.1.3/.configured_68b329da9893e34099c7d8ad5cb9c940' failed
make[3]: *** [/home/ling/lede/build_dir/target-x86_64_musl/qBittorrent-4.1.3/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[3]: Leaving directory '/home/ling/lede/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#0.21#0.07#0.73
package/Makefile:107: recipe for target 'package/lean/qBittorrent/compile' failed
make[2]: *** [package/lean/qBittorrent/compile] Error 2
make[2]: Leaving directory '/home/ling/lede'
package/Makefile:103: recipe for target '/home/ling/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ling/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ling/lede'
/home/ling/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

zxlhhyccc, 2019-06-13:
> QT5 已经安装好测试过，还是不行，貌似提示是少了某个文件。。
> configure: error: Could not find qmake
> Makefile:57: recipe for target [...]

我认为是少了文件，我编译到这里缺少文件:
[image]

yuankongk, 2019-06-15:
一样各种尝试都不行

closed 2019-06-15 by coolsnowwolf