coolsnowwolf/lede #1399: hiwifi4 mt7621重启配置都丢失为什么

xiangfeigao, 2019-06-19:
hiwifi4 mt7621重启配置都丢失为什么？试验了两次，都是这样

qinfeng8848, 2019-06-19:
你刷入的固件名称？

xiangfeigao, 2019-06-20:
6.1版本的factory

GHSY123, 2019-06-21:
> hiwifi4 mt7621重启配置都丢失为什么？试验了两次，都是这样

生成的两个固件，一个是kernel，另一个是sys upgrade，现在breed管理界面刷入kernel固件，然后进入路由器设置界面在更新备份那里边选择不保留配置刷入sysupgrade那个固件，重启后再进入这个界面重置路由器，至此刷机才算完成，这样就不会每次断电或重启自动恢复配置。

xiangfeigao, 2019-06-22:
大雕已经更新了，重新编译生成，问题解决了

uoow, 2019-06-28:
重新编译能解决吗？用最新代码编译后没有了factory，只有kernel和sysupgrade的固件，这正不正常？而且，用sysupgrade升级后，固件起不来，看网卡是不停的连接断开连接断开。

GHSY123, 2019-06-29:
> 重新编译能解决吗？用最新代码编译后没有了factory，只有kernel和sysupgrade的固件，这正不正常？而且，用sysupgrade升级后，固件起不来，看网卡是不停的连接断开连接断开。

我的也是生成这两个固件。
生成的两个固件，一个是kernel，另一个是sys upgrade，现在breed管理界面刷入kernel固件，然后进入路由器设置界面在更新备份那里边选择不保留配置刷入sysupgrade那个固件，重启后再进入这个界面重置路由器，至此刷机才算完成，这样就不会每次断电或重启自动恢复配置。

uoow, 2019-06-29:
> > 重新编译能解决吗？用最新代码编译后没有了factory，只有kernel和sysupgrade的固件，这正不正常？而且，用sysupgrade升级后，固件起不来，看网卡是不停的连接断开连接断开。
> 
> 我的也是生成这两个固件。
> 生成的两个固件，一个是kernel，另一个是sys upgrade，现在breed管理界面刷入kernel固件，然后进入路由器设置界面在更新备份那里边选择不保留配置刷入sysupgrade那个固件，重启后再进入这个界面重置路由器，至此刷机才算完成，这样就不会每次断电或重启自动恢复配置。

试过你的方法，无效。可能是我用的不是breed吧，用的u-boot

closed 2019-10-18 by coolsnowwolf