coolsnowwolf/lede #1406: AR71xx 编译出错 新人新手 多多关照

jeasonliang, 2019-06-22:
系统是debian9  xshell登录了编译 出现如下问题:

CMake Error at cmake_install.cmake:45 (file):
  file INSTALL cannot copy file
  "/home/debian/openwrt/lede/build_dir/host/cmake-3.13.2/Templates/CMakeVSMacros2.vsmacros"
  to
  "/home/debian/openwrt/lede/staging_dir/host/share/cmake-3.13/Templates/CMakeVSMacros2.vsmacros".

Makefile:73: recipe for target 'install' failed
make[4]: *** [install] Error 1
make[4]: Leaving directory '/home/debian/openwrt/lede/build_dir/host/cmake-3.13.2'
Makefile:35: recipe for target '/home/debian/openwrt/lede/staging_dir/host/stamp/.cmake_installed' failed
make[3]: *** [/home/debian/openwrt/lede/staging_dir/host/stamp/.cmake_installed] Error 2
make[3]: Leaving directory '/home/debian/openwrt/lede/tools/cmake'
time: tools/cmake/compile#841.15#80.40#1613.98
tools/Makefile:155: recipe for target 'tools/cmake/compile' failed
make[2]: *** [tools/cmake/compile] Error 2
make[2]: Leaving directory '/home/debian/openwrt/lede'
tools/Makefile:153: recipe for target '/home/debian/openwrt/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynnyy' failed
make[1]: *** [/home/debian/openwrt/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynnyy] Error 2
make[1]: Leaving directory '/home/debian/openwrt/lede'
/home/debian/openwrt/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

大家指导一下!谢谢~!

ope087, 2019-06-24:
你可以用winscp连上,看看 /home/debian/openwrt/lede/build_dir/host/cmake-3.13.2/Templates/CMakeVSMacros2.vsmacros  这个文件有没有,大小是不是62K,然后手动复制到/home/debian/openwrt/lede/staging_dir/host/share/cmake-3.13/Templates/  下.再编译试试,还有你是不是用root用户编译过了?

jeasonliang, 2019-06-25:
感谢～～！回头试试～！

closed 2019-06-28 by coolsnowwolf