coolsnowwolf/lede #1409: 请问下是否可以运行Docker

darktitty, 2019-06-24:
请问下是否可以运行Docker?

ope087, 2019-06-24:
可以,x86_64上我运行过.

darktitty, 2019-06-24:
@ope087 是需要勾选内核之类的编译固件才可以吗? 有参考资料或者可以跟您请教一下吗?

ope087, 2019-06-24:
对的,需要内核的支持,最简单的就是去luci里把lxc勾上,刷好后手动链接文件到bin里,init脚本的启动命令我是抄的koolshare的固件里的,

darktitty, 2019-06-24:
还有后面两步手动链接和init脚本启动啊...这个不太懂啊 好吧 我先研究下看看 不懂再请教你 谢谢

darktitty, 2019-06-25:
@ope087 是不是勾选了lxc以后编译出的固件就无法正常启动了? 刷好后手动链接文件到bin里,init脚本的启动命令我是抄的koolshare的固件里的, 这部分可以稍微详细说明下吗? 谢谢啦

ope087, 2019-06-25:
去https://download.docker.com/linux/static/stable/下载你对应cpu的docker,
解压后把里面的可执行文件全部用ln命令,链接到/usr/bin下面(也可以是其他bin目录下)
以下是init脚本
------------------------------------------
#!/bin/sh /etc/rc.common

START=99

start(){
    /usr/sbin/dockerd --config-file=/mnt/你挂载的磁盘/docker/config/daemon.json --data-root=/mnt/你挂载的磁盘/docker/data --exec-root=/mnt/你挂载的磁盘/docker/data/run --iptables=true --ip-masq=true --selinux-enabled=false --ip-forward=true --ip=0.0.0.0 --group root > /dev/null 2>&1 &
	}
	
stop(){
    killall dockerd
	}
	
restart(){
	stop
	start
}

darktitty, 2019-06-25:
@ope087 好的 谢谢 我没搞过这些 我在继续去研究一下 可以给个联系方式吗? 好请教你~

ope087, 2019-06-25:
用户名@qq.com

sqliuchang, 2019-06-30:
> 对的,需要内核的支持,最简单的就是去luci里把lxc勾上,刷好后手动链接文件到bin里,init脚本的启动命令我是抄的koolshare的固件里的,

为啥我的koolshare里面的docker_config.sh是个二进制文件，而且开机自启有问题

closed 2019-10-18 by coolsnowwolf