coolsnowwolf/lede #1410: 自编译的ssrplus中v2ray不太正常

jonsonjordan, 2019-06-24:
用最新源码编译的，解锁了ssrplus。自建的v2ray服务器信息填到ssrplus里运行正常，机场的订阅节点手机上使用正常，用ssrplus能更新订阅，正常显示节点信息，可是使用机场节点后ssrplus状态就一直显示未运行，也不能科学，请问需要编译哪些依赖进去？已附上机场节点的配置
[image]
[image]
[image]
[image]

angelsclare, 2019-06-25:
加密方式是否与服务器匹配？

jonsonjordan, 2019-06-26:
> 加密方式是否与服务器匹配？transport的几个方式多感觉也也没有选择对。

经你提醒，我对比了手机订阅和路由订阅的服务器信息，发现路由订阅导入的配置里丢失了‘websocket host’信息，填上就可以了，算是v2ray订阅功能的一个小bug吧

tty228, 2019-06-27:
我仓库那里有个修复版的

tty228, 2019-06-27:
我仓库上有个修复过的，
[https://github.com/tty228/lede/tree/master/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr](url)

主要是 `subscribe.sh` 这个文件

jonsonjordan, 2019-06-27:
> https://github.com/tty228/lede/tree/master/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr
多谢

closed 2019-10-18 by coolsnowwolf

wang3stone, 2021-03-21:
> > https://github.com/tty228/lede/tree/master/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr
> > 多谢

没有了？
一样的问题， SSR Plus 用不了V2ray

Linkcy97, 2022-11-03:
> 

编译的时候选上v2ray核心