coolsnowwolf/lede #1413: 断网，需要拔掉电源才恢复

ihuaer, 2019-06-26:
MTK7621 er-x路由器，间隙性断网，需要拔掉路由器电源才能恢复正常

MediaTek MT7621 ver:1 eco:3 x 4
--
固件版本 | OpenWrt R9.6.19 / LuCI Master (git-19.146.62144-fd6fdb2)

> ``Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=19438 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=20426 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=21072 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=25013 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=25094 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=26006 not support ','
Wed Jun 26 10:22:50 2019 daemon.err koolproxy[4551]: rule line=26662 not support ','
Wed Jun 26 16:39:19 2019 cron.err crond[2979]: time disparity of 376 minutes detected
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4743]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4745]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4747]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4750]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4752]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4755]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4757]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4759]: initializing ciphers... aes-256-cfb
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4747]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4752]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4745]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4747]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4752]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4745]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4747]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4752]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4745]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4757]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4750]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4757]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4750]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4757]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4750]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4743]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4743]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4743]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4759]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4755]: tcp port reuse enabled
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4759]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4759]: running from root user
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4755]: listening at 0.0.0.0:1234
Wed Jun 26 16:39:29 2019 kern.info /usr/bin/ssr-redir[4755]: running from root user
Wed Jun 26 16:39:29 2019 daemon.info pdnsd[4782]: pdnsd-1.2.9b-par starting.
Wed Jun 26 16:39:30 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jun 26 16:39:30 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jun 26 16:41:47 2019 daemon.err uhttpd[1717]: luci: accepted login on / for root from 192.168.10.234
Wed Jun 26 16:47:41 2019 kern.err /usr/bin/ssr-redir[4759]: server recv: Connection reset by peer
Wed Jun 26 16:50:38 2019 kern.err /usr/bin/ssr-redir[4759]: server recv: Connection reset by peer

choushui007, 2019-06-27:
我的也有这个问题

coolsnowwolf, 2019-06-28:
固件不是我这里的，包含了 koolshare 的东西，概不负责！

closed 2019-10-18 by coolsnowwolf