coolsnowwolf/lede #1414: 编译marvell armdada 3700开发板出现内核驱动错误。

lqhorochi, 2019-06-26:
drivers/net/ethernet/marvell/mvneta.c: In function 'mvneta_probe':
drivers/net/ethernet/marvell/mvneta.c:4467:3: error: label 'err_free_stats' used but not defined
   goto err_free_stats;
   ^~~~
make[9]: *** [drivers/net/ethernet/marvell/mvneta.o] Error 1
make[8]: *** [drivers/net/ethernet/marvell] Error 2
make[7]: *** [drivers/net/ethernet] Error 2
make[6]: *** [drivers/net] Error 2
make[5]: *** [drivers] Error 2
make[5]: Leaving directory `/home/kenneth/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/linux-4.14.127'
make[4]: *** [/home/kenneth/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/linux-4.14.127/.image] Error 2
make[4]: Leaving directory `/home/kenneth/lede/target/linux/mvebu'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/kenneth/lede/target/linux'
time: target/linux/install#264.88#16.60#342.39
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/kenneth/lede'
make[1]: *** [/home/kenneth/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/kenneth/lede'
make: *** [world] 错误 2

lqhorochi, 2019-06-26:
openwrt网:https://openwrt.org/toh/hwdata/globalscale/globalscale_espressobin_v5_0_1
开发板官网:http://espressobin.net/tech-spec/
可以看看

dunkeng, 2019-06-27:
https://bugs.openwrt.org/index.php?do=details&task_id=2322

openwrt已经修复，坐等大雕更新代码

lqhorochi, 2019-06-27:
大雕忙着开车呢~

closed 2019-10-18 by coolsnowwolf