coolsnowwolf/lede #1425: ShadowSocksR Plus+ 的v2ray 不支持mKCP协议，TCP正常。

zxl777, 2019-06-29:
v2ray用mKCP测试不通，没有其他错误提示，折腾半天才发现只有TCP才行。

owen-gu, 2019-06-29:
我可以

GHSY123, 2019-06-29:
支持，我就在用

zxl777, 2019-06-29:
> 支持，我就在用

麻烦分享一下能用的版本号，谢谢。

GHSY123, 2019-06-29:
v4.19-3

发送自 Windows 10 版邮件<https://go.microsoft.com/fwlink/?LinkId=550986>应用

zxlhhyccc, 2019-06-29:
> v4.19-3 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986>%E5%BA%94%E7%94%A8)
> […](#)

你4.19-3哪里可以找到这个版本号？

GHSY123, 2019-06-29:
> > v4.19-3 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986%3E%E5%BA%94%E7%94%A8)
> > […](#)
> 
> 你4.19-3哪里可以找到这个版本号？

软件包搜索v2ray

zxlhhyccc, 2019-06-29:
> > > v4.19-3 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986%3E%E5%BA%94%E7%94%A8)
> > > […](#)
> > 
> > 
> > 你4.19-3哪里可以找到这个版本号？
> 
> 软件包搜索v2ray

github上搜索出来还是4.19-1版本

GHSY123, 2019-06-30:
> > v4.19-3 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986%3E%E5%BA%94%E7%94%A8)
> > […](#)
> 
> 你4.19-3哪里可以找到这个版本号？

用大雕源码编译完之后就是这个版本，内核R9.6.19。

zxlhhyccc, 2019-06-30:
> > > v4.19-3 发送自 Windows 10 版邮件<[https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986%3E%E5%BA%94%E7%94%A8)
> > > […](#)
> > 
> > 
> > 你4.19-3哪里可以找到这个版本号？
> 
> 用大雕源码编译完之后就是这个版本，内核R9.6.19。

大雕的是4.19不是4.19-3吧

hcym, 2019-06-30:
这个一直正常使用的是常规

GHSY123, 2019-07-01:
> > > > v4.19-3发送自窗10版邮件< [https://go.microsoft.com/fwlink/?LinkId=550986>应用](https://go.microsoft.com/fwlink/?LinkId=550986%3E%E5%BA%94%E7%94%A8)
> > > > [...](#) 
> > > 
> > > 
> > > 你4.19-3哪里可以找到这个版本号？
> > 
> > 
> > 用大雕源码编译完之后就是这个版本，内核R9.6.19。
> 
> 大雕的是4.19不是4.19-3吧

我也是无话可说。
[image]

zxl777, 2019-07-01:
版本在此，确实在kcp下不能工作。用的一键脚本。

[image]

owen-gu, 2019-07-02:
> 版本在此，确实在kcp下不能工作。用的一键脚本。
> 
> [image]

估计服务器问题， 我有个VPS封了KCP

qiangwushuang, 2019-08-01:
有没有解决啊？刚更新的，同样是发现路由器kcp方式上不了（经测试直接通过电脑连接v2ray是可以的），版本同样是v4.19-3。

GHSY123, 2019-08-01:
> 有没有解决啊？刚更新的，同样是发现路由器kcp方式上不了（经测试直接通过电脑连接v2ray是可以的），版本同样是v4.19-3。

截个图。

GHSY123, 2019-08-01:
> > > 有没有解决啊？刚更新的，同样是发现路由器kcp方式上不了（经测试直接通过电脑连接v2ray是可以的），版本同样是v4.19-3。
> > 
> > 
> > 截个图。
> 
> [image]
> [image]

我用的是ssr plus里面的v2ray服务，你这个是独立的v2ray组件吗？

closed 2019-10-18 by coolsnowwolf