coolsnowwolf/lede #1428: openwrt 软路由下使用ssr，ps4设备网速问题。

hdzj-1981, 2019-06-30:
软路由是 3216u ，8g ，120g ssd，四网口。。。。。直接安装openwrt，使用不可描述上路，我的ps4pro 的网络测试只有十几Mb/s，下载超级慢，就算把ps4pro 的ip绑定，再排除在不可描述上路之外也不行，关掉这个功能，在测试，ps4pro就能跑到200Mb/s，下载速度也到满速了.换了好几个版本的openwrt了。都是一样的问题。。。。 今天我换k2p 刷的pda ，用同样的不可描述，就一切正常。。。   我吧我能想到的设置都调试了，反复折腾了好多天了。。。始终无法解决。。。特向大神们请教。。。谢谢了。。。。。

OpenWrt R9.5.1 for x86/x64

主机名 | OpenWrt
-- | --
主机型号 | Intel(R) Celeron(R) CPU 3215U @ 1.70GHz : 2 Core 2 Thread
CPU信息 | 1700.103 MHz +47.0°C
固件版本 | OpenWrt R9.5.1 By Lean / LuCI Master (git-19.121.62139-9ee402e)
内核版本 | 4.9.171

openwrt系统日志如下：

Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.682613] pci 0000:03:00.0: reg 0x18: [io  0xc000-0xc01f]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.682628] pci 0000:03:00.0: reg 0x1c: [mem 0xf7120000-0xf7123fff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.682755] pci 0000:03:00.0: PME# supported from D0 D3hot D3cold
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.682798] pci 0000:03:00.0: System wakeup disabled by ACPI
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.699866] pci 0000:00:1c.2: PCI bridge to [bus 03]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714818] pci 0000:00:1c.2:   bridge window [io  0xc000-0xcfff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714821] pci 0000:00:1c.2:   bridge window [mem 0xf7100000-0xf71fffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714933] pci 0000:04:00.0: [8086:1539] type 00 class 0x020000
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714953] pci 0000:04:00.0: reg 0x10: [mem 0xf7000000-0xf701ffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714981] pci 0000:04:00.0: reg 0x18: [io  0xb000-0xb01f]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.714996] pci 0000:04:00.0: reg 0x1c: [mem 0xf7020000-0xf7023fff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.715125] pci 0000:04:00.0: PME# supported from D0 D3hot D3cold
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.715167] pci 0000:04:00.0: System wakeup disabled by ACPI
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.732236] pci 0000:00:1c.3: PCI bridge to [bus 04]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.747186] pci 0000:00:1c.3:   bridge window [io  0xb000-0xbfff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.747189] pci 0000:00:1c.3:   bridge window [mem 0xf7000000-0xf70fffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.747217] pci_bus 0000:00: on NUMA node 0
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.748370] ACPI: PCI Interrupt Link [LNKA] (IRQs 3 4 5 6 10 *11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.769969] ACPI: PCI Interrupt Link [LNKB] (IRQs 3 4 5 6 10 *11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.791546] ACPI: PCI Interrupt Link [LNKC] (IRQs 3 4 5 6 *10 11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.813127] ACPI: PCI Interrupt Link [LNKD] (IRQs 3 4 5 6 *10 11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.834706] ACPI: PCI Interrupt Link [LNKE] (IRQs 3 4 5 6 10 11 12 14 15) *0, disabled.
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.859898] ACPI: PCI Interrupt Link [LNKF] (IRQs 3 4 5 6 10 11 12 14 15) *0, disabled.
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.885067] ACPI: PCI Interrupt Link [LNKG] (IRQs 3 4 *5 6 10 11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.906669] ACPI: PCI Interrupt Link [LNKH] (IRQs 3 4 *5 6 10 11 12 14 15)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.928515] ACPI: Enabled 4 GPEs in block 00 to 7F
Sun Jun 30 22:59:30 2019 kern.notice kernel: [    1.943280] SCSI subsystem initialized
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    1.954653] libata version 3.00 loaded.
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.954657] ACPI: bus type USB registered
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.966768] usbcore: registered new interface driver usbfs
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.983290] usbcore: registered new interface driver hub
Sun Jun 30 22:59:30 2019 kern.info kernel: [    1.999304] usbcore: registered new device driver usb
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.014559] PCI: Using ACPI for IRQ routing
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027165] PCI: pci_cache_line_size set to 64 bytes
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027286] e820: reserve RAM buffer [mem 0x0009d800-0x0009ffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027287] e820: reserve RAM buffer [mem 0xd5f89000-0xd7ffffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027288] e820: reserve RAM buffer [mem 0xdb379000-0xdbffffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027289] e820: reserve RAM buffer [mem 0xdb5ac000-0xdbffffff]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.027291] e820: reserve RAM buffer [mem 0x21f800000-0x21fffffff]
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.027470] hpet0: at MMIO 0xfed00000, IRQs 2, 8, 0, 0, 0, 0, 0, 0
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.046784] hpet0: 8 comparators, 64-bit 14.318180 MHz counter
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.066368] clocksource: Switched to clocksource hpet
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.081659] pnp: PnP ACPI init
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.091123] system 00:00: [io  0x0a00-0x0a2f] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.108939] system 00:00: [io  0x0a30-0x0a3f] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.126747] system 00:00: [io  0x0a40-0x0a4f] has been reserved
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.144561] system 00:00: Plug and Play ACPI device, IDs PNP0c02 (active)
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.144840] pnp 00:01: [dma 0 disabled]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.144893] pnp 00:01: Plug and Play ACPI device, IDs PNP0501 (active)
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.145130] pnp 00:02: [dma 0 disabled]
Sun Jun 30 22:59:30 2019 kern.debug kernel: [    2.145175] pnp 00:02: Plug and Play ACPI device, IDs PNP0501 (active)
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.145283] system 00:03: [io  0x0680-0x069f] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.163096] system 00:03: [io  0xffff] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.179083] system 00:03: [io  0xffff] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.195074] system 00:03: [io  0xffff] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.211063] system 00:03: [io  0x1800-0x18fe] has been reserved
Sun Jun 30 22:59:30 2019 kern.info kernel: [    2.228875] system 00:03: [io  0x164e-0x164f] has been reserved
[... 1297 lines cut ...]
[   13.634547] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   14.091743] 8021q: adding VLAN 0 to HW filter on device eth1
[   14.109265] br-lan: port 1(eth1) entered blocking state
[   14.125021] br-lan: port 1(eth1) entered disabled state
[   14.140851] device eth1 entered promiscuous mode
[   14.155084] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   14.200027] 8021q: adding VLAN 0 to HW filter on device eth2
[   14.217665] br-lan: port 2(eth2) entered blocking state
[   14.233411] br-lan: port 2(eth2) entered disabled state
[   14.249252] device eth2 entered promiscuous mode
[   14.289930] 8021q: adding VLAN 0 to HW filter on device eth3
[   14.307392] br-lan: port 3(eth3) entered blocking state
[   14.323139] br-lan: port 3(eth3) entered disabled state
[   14.338961] device eth3 entered promiscuous mode
[   14.379636] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   14.397206] 8021q: adding VLAN 0 to HW filter on device eth0
[   17.206808] igb 0000:01:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[   17.466431] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   22.825123] pppoe-wan: renamed from ppp0
[   23.104153] random: fast init done
[   89.740157] random: crng init done
[  178.788655] igb 0000:04:00.0 eth3: igb: eth3 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[  179.048271] br-lan: port 3(eth3) entered blocking state
[  179.064023] br-lan: port 3(eth3) entered forwarding state
[  179.080503] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

sqliuchang, 2019-06-30:
不可描述已经从代码中删除了 大雕估计应该不会维护这个功能了

greekstreet, 2019-07-16:
去etc/dnsmasq.ssr/gfw_base.conf这个文件，把如下行删除，应该就好了。
ipset=/.ac.playstation.net/gfwlist
ipset=/.dl.playstation.net/gfwlist
ipset=/.update.playstation.net/gfwlist
server=/.ac.playstation.net/127.0.0.1#5335
server=/.dl.playstation.net/127.0.0.1#5335
server=/.update.playstation.net/127.0.0.1#5335

hdzj-1981, 2019-07-20:
非常感谢。问题得到了解决。。。。。

closed 2019-10-18 by coolsnowwolf

imvane, 2023-02-19:
> 非常感谢。问题得到了解决。。。。。
> […](#)
请问去哪里删除 我不会配置

imvane, 2023-02-19:
> 

请问去哪里删除