coolsnowwolf/lede #1431: ShadowSocksR Plus+ 数个小时，有时会断流几十秒钟，原因是什么？

zxl777, 2019-07-01:


zxl777, 2019-07-04:
有没有办法看到ShadowSocksR Plus+的log，好判断是什么原因断流？谢谢

tjaol, 2019-07-08:
我好像是换了最新版的 v2ray-core问题消失

zxl777, 2019-07-11:
我现在发现，只要掉线，先断开v2ray，重新连一下马上回复正常，应该只是这个v2ray-core的bug，
不是其他原因。

怎么换v2ray-core？

S0MEW0LF, 2019-07-11:
我用SSR Plus+也有这个现象，使用v2ray协议会有假死现象，需要重新连接v2ray服务器，不然一直连接不上国外网站，从正常到假死时间不定，不过我目前未有超过3天不假死的。使用ss协议就不会，挂着一两个月都不会有问题。

stevebanon, 2019-07-12:
> 我用SSR Plus+也有这个现象，使用v2ray协议会有假死现象，需要重新连接v2ray服务器，不然一直连接不上国外网站，从正常到假死时间不定，不过我目前未有超过3天不假死的。使用ss协议就不会，挂着一两个月都不会有问题。

现在ss还能用？ 什么技术？

S0MEW0LF, 2019-07-13:
> > 我用SSR Plus+也有这个现象，使用v2ray协议会有假死现象，需要重新连接v2ray服务器，不然一直连接不上国外网站，从正常到假死时间不定，不过我目前未有超过3天不假死的。使用ss协议就不会，挂着一两个月都不会有问题。
> 
> 现在ss还能用？ 什么技术？

不清楚，一个用了三四年机场的服务器，6月初也封了三四天，之后一直正常。

chrisqin, 2019-07-14:
> > 我用SSR Plus+也有这个现象，使用v2ray协议会有假死现象，需要重新连接v2ray服务器，不然一直连接不上国外网站，从正常到假死时间不定，不过我目前未有超过3天不假死的。使用ss协议就不会，挂着一两个月都不会有问题。
> 
> 现在ss还能用？ 什么技术？

我一直都在用着SS呀。稳如狗。

closed 2019-08-02 by coolsnowwolf