coolsnowwolf/lede #1436: 编译时出现这个问题，不知道该怎么解决，求帮助

man2018, 2019-07-02:
mv -f .deps/quotearg.Tpo .deps/quotearg.Po
gcc -std=gnu99 -DHAVE_CONFIG_H -I. -I..   -I/home/user1/lede/staging_dir/host/include   -O2 -I/home/user1/lede/staging_dir/host/include  -MT renameat2.o -MD -MP -MF .deps/renameat2.Tpo -c -o renameat2.o renameat2.c
In file included from /usr/include/x86_64-linux-gnu/sys/syscall.h:31:0,
                 from renameat2.c:29:
renameat2.c: In function 'renameat2':
renameat2.c:81:22: error: '__NR_renameat2' undeclared (first use in this function)
   ret_val = syscall (SYS_renameat2, fd1, src, fd2, dst, flags);
                      ^
renameat2.c:81:22: note: each undeclared identifier is reported only once for each function it appears in
make[7]: *** [renameat2.o] Error 1
make[7]: Leaving directory `/home/user1/lede/build_dir/host/patch-2.7.6/lib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/user1/lede/build_dir/host/patch-2.7.6/lib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/user1/lede/build_dir/host/patch-2.7.6'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/user1/lede/build_dir/host/patch-2.7.6'
make[3]: *** [/home/user1/lede/build_dir/host/patch-2.7.6/.built] Error 2
make[3]: Leaving directory `/home/user1/lede/tools/patch'
time: tools/patch/compile#12.37#7.21#24.16
make[2]: *** [tools/patch/compile] Error 2
make[2]: Leaving directory `/home/user1/lede'
make[1]: *** [/home/user1/lede/staging_dir/target-x86_64_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyynyyyyyyyyyyyyyyynyyynynnyyynnyy] Error 2
make[1]: Leaving directory `/home/user1/lede'
make: *** [world] Error 2

Zhang-Shijun, 2019-07-02:
这么跟你说的，其实你报错，发这里几乎没人理你的，那些真正会的大佬，包括lean本人。这东西并没有lean说的那么好，肯定能编译成功，我开着全局梯子，最新源码，该安装的工具安装了，失败了make distclean什么的都做。没有一次成功的，唯一一次成功的，斐讯分区刷进去后都是进不去路由器后台的，换上其他固件，马上没问题。我并没有诋毁lean以及lean的劳动成果的意思，只是实话是说。有问题，自己解决吧，或者等着哪天哪位大佬大发慈悲愿意给你解答。

jackiezhang29, 2019-07-06:
梯子的问题

closed 2019-10-18 by coolsnowwolf