coolsnowwolf/lede #1447: 编译最新 R9.7.6 出错

xuhaojia, 2019-07-07:
scripts/kconfig/conf  --silentoldconfig Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
make[7]: warning:  Clock skew detected.  Your build may be incomplete.
make[6]: warning:  Clock skew detected.  Your build may be incomplete.
make[5]: Leaving directory `/home/xy6538/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.184'
make[5]: Entering directory `/home/xy6538/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.9.184'
make[5]: Warning: File `scripts/Makefile.ubsan' has modification time 3.2e+06 s in the future
make[6]: Warning: File `arch/x86/Makefile' has modification time 3.2e+06 s in the future
make[7]: Warning: File `scripts/Makefile.host' has modification time 3.2e+06 s in the future
  HOSTCC  scripts/basic/fixdep
make[7]: warning:  Clock skew detected.  Your build may be incomplete.
make[7]: Warning: File `scripts/Makefile.host' has modification time 3.2e+06 s in the future
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf

mdzz0, 2019-07-07:
别见怪，这种建议还是先搜一下错误，自己先试试看看能不能解决，看这提示好像挺明确了，，建议可以自己谷歌搜搜处理一下，网上貌似挺多这个问题的

xuhaojia, 2019-07-08:
> 别见怪，这种建议还是先搜一下错误，自己先试试看看能不能解决，看这提示好像挺明确了，，建议可以自己谷歌搜搜处理一下，网上貌似挺多这个问题的

同步时间后解决此问题；
主要是一直以来编译都是正常的。

closed 2019-07-11 by coolsnowwolf