coolsnowwolf/lede #1454: 最新版编译报错

1021683053, 2019-07-09:
```bash
Number of uids 1
	root (0)
Number of gids 1
	root (0)
/home/liweifeng/lede/staging_dir/host/bin/make_ext4fs -l 94371840 -b 4096 -m 0 -J -T 1562423046 /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/root.ext4 /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/root-brcm2708/
Creating filesystem with parameters:
    Size: 94371840
    Block size: 4096
    Blocks per group: 32768
    Inodes per group: 5760
    Inode size: 256
    Journal blocks: 0
    Label:
    Blocks: 23040
    Block groups: 1
    Reserved blocks: 0
    Reserved block group size: 7
Created filesystem with 2518/5760 inodes and 9017/23040 blocks
rm -f /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin
cp /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/vmlinux /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin
perl /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/scripts/mkknlimg /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin.tmp
Version: Linux version 4.14.131 (liweifeng@Liweifeng) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r1417-b072ad4)) #0 Sat Jul 6 14:24:06 2019
DT: y
DDT: y
270x: y
283x: y
mv /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin.tmp /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin
[ -f /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin -a -f /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/root.squashfs ]
rm -f /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot
mkfs.fat -C /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot 16384
mkfs.fat 4.1 (2017-01-24)
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/COPYING.linux ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/bootcode.bin ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/LICENCE.broadcom ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/start.elf ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/start_cd.elf ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/start_x.elf ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/fixup.dat ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/fixup_cd.dat ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/fixup_x.dat ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot cmdline.txt ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot config.txt ::
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/rpi-kernel.bin ::kernel.img
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/arch/arm/boot/dts/bcm2708-rpi-b.dtb ::; mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/arch/arm/boot/dts/bcm2708-rpi-b-plus.dtb ::; mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/arch/arm/boot/dts/bcm2708-rpi-cm.dtb ::; mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/arch/arm/boot/dts/bcm2708-rpi-0-w.dtb ::;
mmd -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot ::/overlays
mcopy -i /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz.boot /home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/linux-4.14.131/arch/arm/boot/dts/overlays/*.dtbo ::/overlays/
Disk full
No free cluster 5 8168
buffer_flush: write: No space left on device
No free cluster 0 8168
make[5]: *** [/home/liweifeng/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/linux-brcm2708_bcm2708/tmp/openwrt-brcm2708-bcm2708-rpi-squashfs-factory.img.gz] Error 1
make[5]: Leaving directory `/home/liweifeng/lede/target/linux/brcm2708/image'
make[4]: *** [install] Error 2
make[4]: Leaving directory `/home/liweifeng/lede/target/linux/brcm2708'
make[3]: *** [install] Error 2
make[3]: Leaving directory `/home/liweifeng/lede/target/linux'
time: target/linux/install#343.67#42.74#390.05
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory `/home/liweifeng/lede'
make[1]: *** [/home/liweifeng/lede/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/stamp/.target_install] Error 2
make[1]: Leaving directory `/home/liweifeng/lede'
make: *** [world] Error 2
```

1021683053, 2019-07-09:
使用 Ubuntu 14 与 18都一样相同报错。

BROBIRD, 2019-07-09:
`buffer_flush: write: No space left on device`
问题很明显

1021683053, 2019-07-09:
@BROBIRD No space left on device 什么意思

1021683053, 2019-07-09:
@BROBIRD 磁盘空间不足？
```bash
Exportable Squashfs 4.0 filesystem, xz compressed, data block size 262144
	compressed data, compressed metadata, compressed fragments, no xattrs
	duplicates are removed
Filesystem size 11696.36 Kbytes (11.42 Mbytes)
	41.97% of uncompressed filesystem size (27866.20 Kbytes)
Inode table size 18846 bytes (18.40 Kbytes)
	22.07% of uncompressed inode table size (85387 bytes)
Directory table size 26316 bytes (25.70 Kbytes)
	45.15% of uncompressed directory table size (58284 bytes)
```

1021683053, 2019-07-09:
@BROBIRD 
```base
Filesystem                     1GB-blocks  Used Available Use% Mounted on
/dev/mapper/Liweifeng--vg-root       67GB  18GB      45GB  29% /
none                                  1GB   0GB       1GB   0% /sys/fs/cgroup
udev                                  1GB   1GB       1GB   1% /dev
tmpfs                                 1GB   1GB       1GB   1% /run
none                                  1GB   0GB       1GB   0% /run/lock
none                                  1GB   1GB       1GB   1% /run/shm
none                                  1GB   0GB       1GB   0% /run/user
/dev/sda1                             1GB   1GB       1GB  19% /boot
/home/liweifeng/.Private             67GB  18GB      45GB  29% /home/liweifeng
```

1021683053, 2019-07-09:
### 使用服务器编译

[image]

[image]

1021683053, 2019-07-10:
@coolsnowwolf 这个是最新版问题吗？之前打成功过。

lifetyper, 2019-07-10:
你就不能看看你编译机器的硬盘到底还有没有空间？

ihuaer, 2019-07-10:
硬盘空间不够了啊

1021683053, 2019-07-10:
@lifetyper 我上面标注了，剩余空间
我图已经截出来了，剩余空间充足的。

1021683053, 2019-07-10:
@ihuaer 看截图，我剩余空间充足的。

alexyw, 2019-07-10:
我也遇到同样的问题，很奇怪

lwjef, 2019-07-10:
服务器重启一下

snowrice, 2019-07-10:
我也是用snowwolf最新源碼編譯時出相同問題

1021683053, 2019-07-11:
问题解决了：
选择问题，之前我选择的
bcm2708->bcm2708(这个是老树莓派设备) 这个有问题
bcm2708->bcm2710(这个是树莓派3与3b的设备) 这个没问题，编译通过

closed 2019-07-11 by 1021683053

snowrice, 2019-07-11:
> ### 问题解决了：
> 选择问题，之前我选择的
> bcm2708->bcm2708(这个是老树莓派设备) 这个有问题
> bcm2708->bcm2710(这个是树莓派3与3b的设备) 这个没问题，编译通过

請問如果用2B2B能用3或3b的固件嗎?

1021683053, 2019-07-11:
@snowrice 很明显，不能，换个树莓派3B吧。

fujie-xiyou, 2019-07-12:
老版本树莓派不配拥有姓名吗？哭了