coolsnowwolf/lede #1456: wndr4300v1 building error

ddddckq, 2019-07-11:
全程科学上网环境下编译，报如下错误，求助大神帮忙看下如何结局？

make[5]: Entering directory '/home/lg/downloads/src/lede/build_dir/host/e2fsprogs-1.44.5/e2fsck'
        MKDIR_P /home/lg/downloads/src/lede/staging_dir/host/bin /home/lg/downloads/src/lede/staging_dir/host/share/man/man8
make[5]: config/install-sh: Command not found
Makefile:503: recipe for target 'installdirs' failed
make[5]: *** [installdirs] Error 127
make[5]: Leaving directory '/home/lg/downloads/src/lede/build_dir/host/e2fsprogs-1.44.5/e2fsck'
Makefile:412: recipe for target 'install-progs-recursive' failed
make[4]: *** [install-progs-recursive] Error 1
make[4]: Leaving directory '/home/lg/downloads/src/lede/build_dir/host/e2fsprogs-1.44.5'
Makefile:53: recipe for target '/home/lg/downloads/src/lede/staging_dir/host/stamp/.e2fsprogs_installed' failed
make[3]: *** [/home/lg/downloads/src/lede/staging_dir/host/stamp/.e2fsprogs_installed] Error 2
make[3]: Leaving directory '/home/lg/downloads/src/lede/tools/e2fsprogs'
time: tools/e2fsprogs/compile#0.14#2.09#2.09
tools/Makefile:155: recipe for target 'tools/e2fsprogs/compile' failed
make[2]: *** [tools/e2fsprogs/compile] Error 2
make[2]: Leaving directory '/home/lg/downloads/src/lede'
tools/Makefile:153: recipe for target '/home/lg/downloads/src/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynnyy' failed
make[1]: *** [/home/lg/downloads/src/lede/staging_dir/target-mips_24kc_musl/stamp/.tools_compile_yynyyyyynyyyyynyynnnyyyyyyyyyyyyyyyyyyyynnyynynyyyynnyy] Error 2
make[1]: Leaving directory '/home/lg/downloads/src/lede'
/home/lg/downloads/src/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

ddddckq, 2019-07-13:
solved by manually correct the Makefile under e2fsprogs sub folder.

closed 2019-07-13 by ddddckq

touxiaoling, 2020-11-09:
> solved by manually correct the Makefile under e2fsprogs sub folder.

你好,请问是如何修改的?我也遇到了同样的问题.
感谢.