coolsnowwolf/lede #1457: fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD

zaoqi, 2019-07-11:


closed 2019-07-26 by coolsnowwolf

fanck0605, 2020-04-01:
同步源码的时候我也遇到了。全局代理环境。

rxchen, 2020-04-03:
Run cd openwrt
2 thread compile
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..9a65c3ef74295054aced5fcbf06bd64d1906e740
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..9a65c3ef74295054aced5fcbf06bd64d1906e740
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..HEAD
fatal: Invalid revision range 97a4ffcc125611dd5f307d54570373832a73e62d..9a65c3ef74295054aced5fcbf06bd64d1906e740

我编译也碰到同样的问题，刚刚通过actions编译的