coolsnowwolf/lede #1468: ShadowSocksR Plus+ 中的kcptun要怎么设置？？？？？？？

aqswaqsw, 2019-07-13:
研究好久没弄明白，给个例子呗

chrisqin, 2019-07-14:
你首先要搞清楚服务器端有没有开kcptun。

ntlf9t, 2019-07-14:
【KCPTUN使用说明】
1、KCPTUN与S-S R绑定，与S-S R服务器一起配置，KCPTUN必须和S-S R服务器运行于同一台服务器
2、配置举例，假设使用SS服务器，服务器IP为11.22.33.44，端口为8388，S-S R密码为S-S Rpass；KCPTUN端口为4000，密码为kcppass，非压缩方式，模式为fast2
在界面上按下图进行配置：
  
kcp参数可以自己填写，以空格隔开，参数说明请点击图上的【参考】链接；注意：参数一般以“--”开头，不是“-”--crypt none --mode fast --mtu 1350 --sndwnd 512 --rcvwnd 512 --datashard 10  --parityshard 3 --dscp 0 --nocomp true   参考请按照自己实际参数来

sxml, 2019-09-01:
ShadowsocksR plus+插件 怎么界面里kcptun才端口和密码 kcptun其他参数不能设置？

angrysad, 2019-10-05:
> 【KCPTUN使用说明】
> 1、KCPTUN与S-S R绑定，与S-S R服务器一起配置，KCPTUN必须和S-S R服务器运行于同一台服务器
> 2、配置举例，假设使用SS服务器，服务器IP为11.22.33.44，端口为8388，S-S R密码为S-S Rpass；KCPTUN端口为4000，密码为kcppass，非压缩方式，模式为fast2
> 在界面上按下图进行配置： [...]

大佬，有个问题，一个多月了还是没搞明白，关于kcptun设置的，望指导。

closed 2019-10-18 by coolsnowwolf