coolsnowwolf/lede #1470: v2ray-core已更新4.20.0,麻烦大雕更新一下SSR Plus+插件

lqhorochi, 2019-07-15:
v2ray-core已更新4.20.0,麻烦大雕更新一下SSR Plus+
https://github.com/v2ray/v2ray-core/releases

zcluo, 2019-07-16:
自己更不就好了

发自我的 iPhone

tefiszx, 2019-07-16:
我是直接从github下载编译好的二进制文件，替换路由器上的版本

mrclock, 2019-07-16:
X86的话直接更新就行啦。

wget https://install.direct/go.sh
bash go.sh

jonsonjordan, 2019-07-16:
> 我是直接从github下载编译好的二进制文件，替换路由器上的版本

X86应该下载哪个版本的二进制文件？

mrclock, 2019-07-16:
> > 我是直接从github下载编译好的二进制文件，替换路由器上的版本
> 
> X86应该下载哪个版本的二进制文件？

就用我上面的两行命令执行脚本就行啦，脚本自动识别系统版本的，直接更新。不用二进制文件。

需要curl和unzip，

opkg update
opkg install curl unzip
wget https://install.direct/go.sh
bash go.sh

就行啦。

jonsonjordan, 2019-07-16:
> > > 我是直接从github下载编译好的二进制文件，替换路由器上的版本
> > 
> > 
> > X86应该下载哪个版本的二进制文件？
> 
> 就用我上面的两行命令执行脚本就行啦，脚本自动识别系统版本的，直接更新。不用二进制文件。
> 
> 需要curl和unzip，
> 
> opkg update
> opkg install curl unzip
> wget https://install.direct/go.sh
> bash go.sh
> 
> 就行啦。

我看到你提到的这个脚本了，运行提示没有curl，安装curl失败，提示如下：
Installing curl (7.65.1-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/base/curl_7.65.1-1_x86_64.ipk
Installing libmbedtls12 (2.16.2-1) to root...
Downloading https://openwrt.proxy.ustclug.org/snapshots/packages/x86_64/base/libmbedtls12_2.16.2-1_x86_64.ipk
Collected errors:
 * check_data_file_clashes: Package libmbedtls12 wants to install file /usr/lib/libmbedcrypto.so.3
	But that file is already provided by package  * libmbedtls
 * check_data_file_clashes: Package libmbedtls12 wants to install file /usr/lib/libmbedtls.so.12
	But that file is already provided by package  * libmbedtls
 * check_data_file_clashes: Package libmbedtls12 wants to install file /usr/lib/libmbedx509.so.0
	But that file is already provided by package  * libmbedtls
 * opkg_install_cmd: Cannot install package curl.

owen-gu, 2019-07-17:
直接替换启不动， v2ray文件要10MB+，

owen-gu, 2019-07-17:
直接替换日子， 实际没有启动。 

[image]

jonsonjordan, 2019-07-17:
> 直接替换日志， 实际没有启动。 提示shadowsocksR plus 未运行
> 
> [image]

一脸懵，我的没有这个情况，你试试用上面那个脚本替换

ywgf88, 2019-07-21:
更新后，在哪个地方看版本？

mrclock, 2019-07-22:
> 更新后，在哪个地方看版本？

最偷懒的办法。再执行下更新，能看到当前版本号是多少，如果与最新版本相同是不会继续更新的。

tefiszx, 2019-07-22:
直接ssh连入路由器，运行一下命令不就能看到版本了。
/usr/local/bin/v2ray -version

AmibaWorld, 2019-07-23:
自己编译更替，是否有压缩，哪位指导下啊，具体如何编译压缩

cnekol, 2019-07-24:
> > > > 我是直接从github下载编译好的二进制文件，替换路由器上的版本
> > > 
> > > 
> > > X86应该下载哪个版本的二进制文件？
> > 
> > 
> > 就用我上面的两行命令执行脚本就行啦，脚本自动识别系统版本的，直接更新。不用二进制文件。
> > 需要curl和unzip，
> > opkg update
> > opkg install curl unzip
> > wget https://install.direct/go.sh
> > bash go.sh
> > 就行啦。
> 
> 我看到你提到的这个脚本了，第一次ssh安装curl失败我以为不能用，结果luci装curl成功了，已更新，谢谢

兄台，我ssh，网页里lede两种方式返回的都是同一种结果。说无法安装。
你是怎么做到安装curl的？lede中的包选择的哪一个？

mrclock, 2019-07-24:
> 兄台，我ssh，网页里lede两种方式返回的都是同一种结果。说无法安装。
> 你是怎么做到安装curl的？lede中的包选择的哪一个？

SSH下看你的错误提示啊，如果是什么文件已存在，直接rm掉就行啦。

cnekol, 2019-07-24:
> SSH下看你的错误提示啊，如果是什么文件已存在，直接rm掉就行啦。

不是什么文件已存在，直接是未知包，界面里lede跟ssh里一样。opkg update时你们所有仓库都ok吗？我这里update时，会有仓库说签名没过。

以下是opkg install curl的结果

Unknown package 'curl'.
Collected errors:
 * opkg_install_cmd: Cannot install package curl.

jonsonjordan, 2019-07-25:
> 不是什么文件已存在，直接是未知包，界面里lede跟ssh里一样。opkg update时你们所有仓库都ok吗？我这里update时，会有仓库说签名没过。
>
> 以下是opkg install curl的结果
>
> Unknown package 'curl'.
> Collected errors:
>
> * opkg_install_cmd: Cannot install package curl.

opkg update有错误的话可能是你编译环境不够科学

cnekol, 2019-07-25:
> > > > > > > > 我是直接从github下载编译好的二进制文件，替换路由器上的版本
> > > > > > > 
> > > > > > > 
> > > > > > > X86应该下载哪个版本的二进制文件？
> > > > > > 
> > > > > > 
> > > > > > 就用我上面的两行命令执行脚本就行啦，脚本自动识别系统版本的，直接更新。不用二进制文件。
> > > > > > 需要curl和unzip，
> > > > > > opkg update
> > > > > > opkg install curl unzip
> > > > > > wget https://install.direct/go.sh
> > > > > > bash go.sh
> > > > > > 就行啦。
> > > > > 
> > > > > 
> > > > > 我看到你提到的这个脚本了，第一次ssh安装curl失败我以为不能用，结果luci装curl成功了，已更新，谢谢
> > > > 
> > > > 
> > > > 兄台，我ssh，网页里lede两种方式返回的都是同一种结果。说无法安装。
> > > > 你是怎么做到安装curl的？lede中的包选择的哪一个？
> > > 
> > > 
> > > SSH下看你的错误提示啊，如果是什么文件已存在，直接rm掉就行啦。
> > 
> > 
> > 不是什么文件已存在，直接是未知包，界面里lede跟ssh里一样。opkg update时你们所有仓库都ok吗？我这里update时，会有仓库说签名没过。
> > 以下是opkg install curl的结果
> > Unknown package 'curl'.
> > Collected errors:
> > 
> > * opkg_install_cmd: Cannot install package curl.
> 
> opkg update有错误的话可能是你编译环境不够科学。第一次我ssh安装curl失败，具体什么错误提示忘了，然后到luci里搜到curl这个包点安装就成功了

先前提示是签名检查没通过。今早试了下又可以了。可能网络问题吧。现在已经可以通过opkg安装了。

coolsnowwolf, 2019-07-26:
[image]

coolsnowwolf, 2019-07-26:
暂时更新的意义不大

因为我用的 v2ray 主程序是自己定制和压缩的，搞起来很麻烦，等下个版本

ntgeralt, 2019-08-09:
请问QCA9558路由的ssr plus带的v2ray是4.14版，如果只想更新这插件，请教大概需要怎么做

closed 2019-10-18 by coolsnowwolf

Rangtian, 2020-06-27:
@coolsnowwolf L大，请问 ssr plus+插件想要更新v2ray，是不是就只能重新编译了？有没有直接在线升级的办法？

最近 v2ray 更新很频繁，还都是重要更新，看着 ssr plus+里的 v2ray 还停留在旧版，有点蓝受。