coolsnowwolf/lede #1473: 问下，总是自动就断网重连是什么问题？内容里是系统日志

aqswaqsw, 2019-07-16:
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:14:45 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:15:09 2019 daemon.info pppd[15018]: No response to 5 echo-requests
Wed Jul 17 06:15:09 2019 daemon.notice pppd[15018]: Serial link appears to be disconnected.
Wed Jul 17 06:15:10 2019 daemon.info pppd[15018]: Connect time 19.7 minutes.
Wed Jul 17 06:15:10 2019 daemon.info pppd[15018]: Sent 326152314 bytes, received 2725783560 bytes.
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: Failed to get IP for interface pppoe-wan
Wed Jul 17 06:15:10 2019 daemon.warn miniupnpd[2128]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed Jul 17 06:15:10 2019 daemon.err miniupnpd[2128]: ioctl(s, SIOCGIFADDR, ...): Address not available
Wed Jul 17 06:15:10 2019 daemon.err odhcp6c[19149]: Failed to send RS (Permission denied)
Wed Jul 17 06:15:10 2019 daemon.err odhcp6c[19149]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Network device 'pppoe-wan' link is down
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Network alias 'pppoe-wan' link is down
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Interface 'wan_6' has link connectivity loss
Wed Jul 17 06:15:10 2019 daemon.err odhcp6c[19149]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Interface 'wan' has lost the connection
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Interface 'wan_6' is now down
Wed Jul 17 06:15:10 2019 daemon.notice netifd: Interface 'wan_6' is disabled
Wed Jul 17 06:15:10 2019 daemon.notice pppd[15018]: Connection terminated.
Wed Jul 17 06:15:10 2019 daemon.info pppd[15018]: Sent PADT
Wed Jul 17 06:15:10 2019 daemon.info pppd[15018]: Exit.
Wed Jul 17 06:15:11 2019 daemon.notice netifd: Interface 'wan' is now down
Wed Jul 17 06:15:11 2019 daemon.notice netifd: Interface 'wan' is setting up now
Wed Jul 17 06:15:11 2019 daemon.err insmod: module is already loaded - slhc
Wed Jul 17 06:15:11 2019 daemon.err insmod: module is already loaded - ppp_generic
Wed Jul 17 06:15:11 2019 daemon.err insmod: module is already loaded - pppox
Wed Jul 17 06:15:11 2019 daemon.err insmod: module is already loaded - pppoe
Wed Jul 17 06:15:11 2019 daemon.notice netifd: wan (23620): uci: Entry not found
Wed Jul 17 06:15:11 2019 daemon.info pppd[23647]: Plugin rp-pppoe.so loaded.
Wed Jul 17 06:15:12 2019 daemon.info pppd[23647]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Wed Jul 17 06:15:12 2019 daemon.notice pppd[23647]: pppd 2.4.7 started by root, uid 0
Wed Jul 17 06:15:12 2019 daemon.info pppd[23647]: PPP session is 8236
Wed Jul 17 06:15:12 2019 daemon.warn pppd[23647]: Connected to 78:1d:ba:cc:0e:47 via interface eth0.2
Wed Jul 17 06:15:12 2019 kern.info kernel: [10692.181969] pppoe-wan: renamed from ppp0
Wed Jul 17 06:15:12 2019 daemon.info pppd[23647]: Using interface pppoe-wan
Wed Jul 17 06:15:12 2019 daemon.notice pppd[23647]: Connect: pppoe-wan <--> eth0.2
Wed Jul 17 06:15:15 2019 daemon.info pppd[23647]: syncppp not active
Wed Jul 17 06:15:15 2019 daemon.info pppd[23647]: Remote message: Authentication success,Welcome!
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: PAP authentication succeeded
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: peer from calling number 78:1D:BA:CC:0E:47 authorized
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: local  LL address fe80::d519:2e7e:e3ee:4fa5
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: remote LL address fe80::7a1d:baff:fecc:0e47
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: local  IP address 122.157.147.205
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: remote IP address 122.157.146.1
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: primary   DNS address 202.97.224.69
Wed Jul 17 06:15:15 2019 daemon.notice pppd[23647]: secondary DNS address 202.97.224.68
Wed Jul 17 06:15:15 2019 daemon.notice netifd: Network device 'pppoe-wan' link is up
Wed Jul 17 06:15:15 2019 daemon.notice netifd: Interface 'wan' is now up
Wed Jul 17 06:15:15 2019 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Wed Jul 17 06:15:19 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:15:21 2019 user.notice firewall: Reloading firewall due to ifupdate of wan (pppoe-wan)
Wed Jul 17 06:15:23 2019 user.notice ddns-scripts[24357]: myddns_ipv4: PID '24357' started at 2019-07-17 06:15
Wed Jul 17 06:15:31 2019 user.warn ddns-scripts[24357]: myddns_ipv4: Service section disabled! - TERMINATE
Wed Jul 17 06:15:32 2019 user.warn ddns-scripts[24357]: myddns_ipv4: PID '24357' exit WITH ERROR '1' at 2019-07-17 06:15
Wed Jul 17 06:16:16 2019 cron.info crond[26024]: crond (busybox 1.30.0) started, log level 8
Wed Jul 17 06:16:16 2019 cron.err crond[26024]: user root: parse error at 7
Wed Jul 17 06:16:18 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 17 06:16:18 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 17 06:16:18 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 17 06:16:18 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 17 06:16:28 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 17 06:16:28 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 17 06:16:39 2019 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Jul 17 06:16:39 2019 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Jul 17 06:16:55 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:16:55 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:17:00 2019 cron.err crond[26024]: user root: parse error at 7
Wed Jul 17 06:17:35 2019 daemon.notice netifd: Network alias 'pppoe-wan' link is up
Wed Jul 17 06:17:35 2019 daemon.notice netifd: Interface 'wan_6' is enabled
Wed Jul 17 06:17:35 2019 daemon.notice netifd: Interface 'wan_6' has link connectivity
Wed Jul 17 06:17:35 2019 daemon.notice netifd: Interface 'wan_6' is setting up now
Wed Jul 17 06:17:38 2019 daemon.notice netifd: Interface 'wan_6' is now up
Wed Jul 17 06:17:40 2019 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Wed Jul 17 06:18:00 2019 cron.err crond[26024]: user root: parse error at 7
Wed Jul 17 06:22:10 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:25:47 2019 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Wed Jul 17 06:25:59 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:29:38 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:29:54 2019 daemon.err uhttpd[2368]: luci: accepted login on / for root from 192.168.1.235
Wed Jul 17 06:31:09 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer
Wed Jul 17 06:31:09 2019 kern.err /usr/bin/ssr-redir[17716]: server recv: Connection reset by peer

jonsonjordan, 2019-07-17:
路由wan口连接有问题，测试不通过路由器，直接由电脑拨号看效果，我猜可能出问题的位置：进户线，光猫wan口，光猫lan口，路由wan口，连接猫和路由的网线

aqswaqsw, 2019-07-18:
哦，我去试试

closed 2019-07-26 by coolsnowwolf

tufu9441, 2019-08-26:
@aqswaqsw 请问后来解决了吗？

aqswaqsw, 2019-08-26:
算是解决了，开始的时候是一启动bitcomet比特彗星就会路由卡死重启，现在启动时不卡死重启了，变成用着下载时会ping超级高，15k多

tufu9441, 2019-08-26:
@aqswaqsw 日志里还会有ssr-redir[17716]: server recv: Connection reset by peer的错误提示吗？

s1sunny, 2019-09-13:
> 算是解决了，开始的时候是一启动bitcomet比特彗星就会路由卡死重启，现在启动时不卡死重启了，变成用着下载时会ping超级高，15k多

能请教一下如何解决吗？