coolsnowwolf/lede #1474: 最新代码mt7615e驱动使用问题

GHSY123, 2019-07-17:
lean更新的最新代码集成了mt7615e驱动，但是在编译斐讯K2p完成后，发现无线功能一直无法开启，提示无线未开启或未关联，而且配置无线的密码选项里也没有wpa加密方式，不知道是什么问题。这该如何解决？
[image]
[image]

hcym, 2019-07-17:
还是个半成品，用用老毛子算了

laomao9000, 2019-07-18:
kernel-mod，wifi driver，增加选择mt7615e即可

GHSY123, 2019-07-18:
> kernel-mod，wifi driver，增加选择mt7615e即可

我这就是选过的，就是图中的提示无线未开启或关联。

martin0607, 2019-07-18:
我也是 而且只有5G没有2.4G

GHSY123, 2019-07-19:
> 我也是 而且只有5G没有2.4G

对，无线选项那只有一项，设置完后就会显示无线未关联或未开启。你解决没有？

coolsnowwolf, 2019-07-26:
7615dbdc 不能用的

laomao9000, 2019-07-26:
反复测试了下，发现做个扫描，然后添加一个中继，即可激活wifi信号。可以试试。

GHSY123, 2019-07-27:
> 反复测试了下，发现做个扫描，然后添加一个中继，即可激活wifi信号。可以试试。

入门小白，可有具体操作步骤？

admi-229133, 2019-07-29:
> kernel-mod，wifi driver，增加选择mt7615e即可

那怎么取消无线不要把它编译进去固件里面？

GHSY123, 2019-07-29:
> > kernel-mod，wifi driver，增加选择mt7615e即可
> 
> 那怎么取消无线不要把它编译进去固件里面？

把选项取消就行了。

closed 2019-08-02 by coolsnowwolf