coolsnowwolf/lede #1476: ssr+ 和端口转发冲突

windyin, 2019-07-18:
只要一打开SSR Plus + ,  openwrt上设置的端口转发就失效了，  关闭SSR Plus+ 就能恢复。很奇怪
iptables NAT表如下

[image]
[image]

zxlhhyccc, 2019-07-18:
换端口。。。。

windyin, 2019-07-20:
> 换端口。。。。

换什么端口？ 默认的1234？

coolsnowwolf, 2019-07-26:
具体是哪些端口？

aqswaqsw, 2019-07-26:
我也想知道换哪些端口

mrclock, 2019-08-06:
貌似不是端口转发失效。而是V2转发在系统转发前执行了。你用绕过国内IP试试看。你的转发就生效了。然后你到服务器上去看。不用绕过国内ip的话，会有大量的连接日志。

mrclock, 2019-08-06:
> 具体是哪些端口？

大雕，我的情况是，即使这个端口在转发没被配置。比如1433，这个端口扫的人很多，只要外网有人扫这个端口，V2服务器上就会产生大量连接日志(用绕开国内IP不会出现)。也就是访问路由外网ip任一端口都会转发到V2的服务器上，不管是否被配置了转发。

我是X86的。

mrclock, 2019-08-06:
我先执行下，看看。
iptables -t nat -F
iptables -t nat -X GLOBAL

zcluo, 2019-08-06:
我好像也有这个问题

发自我的 iPhone

feeler29, 2019-08-20:
同样遇到这个问题

guzhiyong, 2019-09-20:
也遇到同样的问题，打开ssr+以后在内网的设备无法通过公网ip访问路由器，Transmission的51413端口在防火墙里打开了，检测却是关闭

closed 2019-10-18 by coolsnowwolf

zidane114, 2020-02-07:
遇到同样问题，折腾了一整个下午。。只要ssr+一开，所有的端口转发都失效，怎么解？

bilijinshi, 2021-07-19:
> 
> 
> 遇到同样问题，折腾了一整个下午。。只要ssr+一开，所有的端口转发都失效，怎么解？

同样问题，开了ssr+后，被ssr+处理的目标地址，防火墙规则不生效了（原来是匹配MAC地址并REJECT）