coolsnowwolf/lede #1485: K3 R9.7.6 无法识别移动硬盘

cody1996, 2019-07-22:
连接移动硬盘后未识别到设备，内核日志如下：
```
[  340.710140] usb 4-1: new SuperSpeed USB device number 5 using xhci-hcd
[  340.742535] xhci-hcd 18023000.xhci: ERROR: unexpected command completion code 0x11.
[  340.790234] uas: probe of 4-1:1.0 failed with error -12
```
使用`lsusb -t`查看结果如下：(`Driver`字段为空)
```
root@CODY-K3:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 5, If 0, Class=Mass Storage, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M
```
我两个移动硬盘都是一样的情况，但是我的两个U盘（一个USB3.0，一个USB2.0）都可以正常识别并自动挂载，USB3.0的U盘连接日志如下：
```
[  299.270139] usb 4-1: new SuperSpeed USB device number 4 using xhci-hcd
[  299.301184] usb-storage 4-1:1.0: USB Mass Storage device detected
[  299.309354] scsi host0: usb-storage 4-1:1.0
[  300.320434] scsi 0:0:0:0: Direct-Access     64G      CHIPFANCIER      1.00 PQ: 0 ANSI: 6
[  300.329530] scsi 0:0:0:1: CD-ROM            64G      CD-ROM on Flash  1.00 PQ: 0 ANSI: 6
[  300.338207] sd 0:0:0:0: [sda] 108045271 512-byte logical blocks: (55.3 GB/51.5 GiB)
[  300.346305] sd 0:0:0:0: [sda] Write Protect is off
[  300.351111] sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
[  300.356381] sd 0:0:0:0: [sda] Write cache: disabled, read cache: disabled, doesn't support DPO or FUA
[  300.370511] sd 0:0:0:0: [sda] Attached SCSI removable disk
[  300.432334] ntfs: volume version 3.1.
```
使用`lsusb -t`查看如下：
```
root@CODY-K3:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M
```
可以看到驱动并不是`uas`，所以我怀疑是UAS驱动的问题，但找不到解决的办法。
另有使用AIDA64查看的设备信息，首先是移动硬盘：

> 设备属性	
> 设备描述	USB Attached SCSI (UAS) 大容量存储设备
> 设备 ID 	174C-55AA
> 设备类别	08 / 06 (Mass Storage)
> 设备协议	50
> 修订	0100h
> 制造商	asmedia
> 产品	ASM1153E
> 支持的 USB 版本	3.00
> 当前速度	Super  (USB 3.0)

然后是另一个移动硬盘：

> 设备属性	
> 设备描述	USB Attached SCSI (UAS) 大容量存储设备
> 设备 ID 	2109-0715
> 设备类别	08 / 06 (Mass Storage)
> 设备协议	50
> 修订	E000h
> 制造商	VLI Manufacture String
> 产品	VLI Product String
> 支持的 USB 版本	3.00
> 当前速度	Super  (USB 3.0)

最后是U盘：

> 设备属性	
> 设备描述	USB 大容量存储设备
> 设备 ID 	1F75-0903
> 设备类别	08 / 06 (Mass Storage)
> 设备协议	50
> 修订	0001h
> 制造商	64G
> 产品	CHIPFANCIER
> 支持的 USB 版本	3.00
> 当前速度	Super  (USB 3.0)

可以看到`设备描述`字段两个移动硬盘是相同的，但和U盘并不相同，让我更加怀疑是UAS驱动的问题。
我已经自己搞了两天了，无奈水平有限，走到这一步就完全无从下手了，还请各位大佬帮忙看看。

cody1996, 2019-07-25:
找到一个解决办法，卸载`kmod-usb-storage-uas`驱动，系统会退而使用`kmod-usb-storage`驱动，就可以正常挂载了。
看起来应该是UAS驱动存在bug，这样用不了UAS可能性能会有所损失，但是我试了openwrt官方仓库最新的系统和UAS驱动也是一样的结果，所以暂时也只好如此了。

closed 2019-07-25 by cody1996

nftbty, 2020-03-14:
@cody1996 你好，请问你现在还是一样的情况吗，我现在是R20.3.11，插上移动硬盘不识别，然后日志报错也是一样，然后拿报错内容在issue搜索就找到你这个issue。过了这么久问题还在，建议可以把这个issue重开，让更多人看到，以寻求解决方案，或者无法解决的话，知道为什么无解的原因。

red1412, 2020-04-01:
我也碰到了无法挂载。。。K3编译还有给问题就是单选了K3但是还是出来一整套其他同型号芯片的固件。。。

nftbty, 2020-04-01:
@red1412  只能去掉uas驱动了，去掉了可以挂载，速度也还可以。没有uas，直接shell写入测试大概有100+MB/s，但是不知道为什么，通过sftp和smb读写就慢了。之前用手机测不到1MB，电脑也只有几兆，后来关掉flow offload里的BBR，速度快些了，但电脑测试smb也只能到40+MB/s，倒是用Nginx搭的文件下载和webdav要快很多，不知道是不是开了sendfile的原因。
另外发现下载宝(MT7621)也不能选uas，网上搜的话也有很多机型选了uas驱动就会有问题，去掉就可以了。

red1412, 2020-04-01:
> @red1412 只能去掉uas驱动了，去掉了可以挂载，速度也还可以。没有uas，直接shell写入测试大概有100+MB/s，但是不知道为什么，通过sftp和smb读写就慢了。之前用手机测不到1MB，电脑也只有几兆，后来关掉flow offload里的BBR，速度快些了，但电脑测试smb也只能到40+MB/s，倒是用Nginx搭的文件下载和webdav要快很多，不知道是不是开了sendfile的原因。
> 另外发现下载宝(MT7621)也不能选uas，网上搜的话也有很多机型选了uas驱动就会有问题，去掉就可以了。

kmod-usb-storage-uas是说这个么？

nftbty, 2020-04-01:
@red1412  就是它

daidaiworm, 2020-06-05:
我也遇到这个问题了。并且删除那个驱动似乎也不行。
比较老的usb3移动硬盘可以自动挂载，比较新的2个uas移动硬盘则不行。
[  103.870188] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
[  103.903561] xhci-hcd 18023000.xhci: ERROR: unexpected command completion code 0x11.
[  103.950255] uas: probe of 4-1:1.0 failed with error -12

# 可以正常挂载
root@OpenWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 005: ID 0bc2:5031 Seagate RSS LLC FreeAgent GoFlex USB 3.0
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 5, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M
# 不能挂载
root@OpenWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 004: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 4, If 0, Class=Mass Storage, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M
# 不能挂载
root@OpenWrt:~# lsusb
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 006: ID 0bc2:ab26 Seagate RSS LLC Backup Plus Slim Portable Drive 1 TB
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@OpenWrt:~# lsusb -t
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 5000M
    |__ Port 1: Dev 6, If 0, Class=Mass Storage, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ohci-platform/2p, 12M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/2p, 480M

bronfo, 2020-07-06:
同样的问题，只能临时解决：
usb-storage quirks=174c:55aa:u
但这样，实测速度只有10M/s多一点，不知道有什么好的解决办法