coolsnowwolf/lede #1504: ZeroTier 1.4版发布了，能更新一下吗？

lifetyper, 2019-07-30:
这个是一个挺大的版本更新，希望可以导入。

lxwq, 2019-07-30:
不知道稳定性怎么样，windows,linux,路由器，都在用这个互通

lifetyper, 2019-07-30:
> 不知道稳定性怎么样，windows,linux,路由器，都在用这个互通

我已经自己更新了，实测下来非常爽！！！！！
一方面是接入网络很快，另一方面传输大文件的时候再也不会掉速了。
以前刚开始的时候速度还行，传输了十几兆之后就掉到几K，现在已经完全没这个问题了。
之前查这个问题的会后，就看到zt的github上有人提这个问题，官方的回复就是1.4会有重大的更新，肯定会解决这个问题，现在真的好了。

lifetyper, 2019-07-30:
> 不知道稳定性怎么样，windows,linux,路由器，都在用这个互通
  我自己fork了大雕的package库，改了zerotier的版本（其实就是个版本号和hash），如果急用可以把feeds.conf.default文件里的大雕的库改成我的，直接编译就行了。

https://github.com/lifetyper/packages

RealKiro, 2019-08-01:
[OP官方版本还在1.2.4，没办法](https://github.com/openwrt/packages)

lifetyper, 2019-08-01:
> [OP官方版本还在1.2.4，没办法](https://github.com/openwrt/packages)

自己改版本和hash值就可以了，我都用上了。

hswmartin, 2019-08-02:
> > [OP官方版本还在1.2.4，没办法](https://github.com/openwrt/packages)
> 
> 自己改版本和hash值就可以了，我都用上了。

实测可以用1.4，不过用你的package库不行，我是直接改大雕的makefile里的hash编译才成功

lifetyper, 2019-08-02:
> > > [OP官方版本还在1.2.4，没办法](https://github.com/openwrt/packages)
> > 
> > 
> > 自己改版本和hash值就可以了，我都用上了。
> 
> 实测可以用1.4，不过用你的package库不行，我是直接改大雕的makefile里的hash编译才成功

不知道你操作有什么特别的，我就是用我自己的package库。

arfaWong, 2019-08-05:
官方版本更新到 1.4.0 了
https://github.com/openwrt/packages/commit/393bb305fcea78cd2e4c3222f9b3a1c36a0a5691

closed 2019-08-05 by coolsnowwolf