coolsnowwolf/lede #1505: MIWIFI ROUTER R3 刷入无法正常启动

hanandjun, 2019-07-30:
您好，首先问一下有哪些可以交流的地方，qq群已禁止加入。

make menuconfig 只是单纯的选了mi router r3，然后编译出来其中有kernel1.bin 和 rootfs0.bin。通过在官方固件下刷入 kernel1 和 rootfs0,启动和下面用 tftp 刷入的一样。

从 u-boot 刷入是把 kernel1.bin 和 rootfs0.bin 拼在了一起。命令是从一篇博客上看到的：

https://blog.ysc3839.com/archives/2018/03/miwifi-r3-ttl-recovery.html

`dd if=/dev/zero bs=4M count=1 | tr "\000" "\377" > padded_kernel.bin`

`dd if=kernel.bin of=padded_kernel.bin conv=notrunc`

`cat padded_kernel.bin rootfs.bin > kernel_rootfs.bin`

下面的输出是从tftp刷入之后打印的

Filename 'kernel_rootfs.bin'.������IMEOUT_COUNT=10,Load address: 0x80100000
Loading����Got ARP REPLY, set server/gtwy �����dr (00:0e:c6:b3:e3:f3)
Got it
Got ARP REQUESԬ����urn our IP
T #�####�#################�########�########�##�###############��##
	 ######�##########�#############�######�############�#�##��######�
	 ########################�###�############��######�###############
	 ############�#�#�################################################
	 �#########�############�###�###�###�###############�#�#####��####
	 ###�#####################��#####�#�####�###�#####��#######�######
	 #####��#####�#�############�###############�######�####�#########
	 ####�#######�#�#############�####�#�#�###########################
	 #��###���####�##�##�####��######�########�##########�######�####�
	 ############################�###�##############################�#
	 ##############�#########�#�#########�############�###############
	 ##�#####�#�#��#################�##########��#####################
	 ##########################��##�#�##########�#�##############�####
	 ######��#################�########�######�####��###��######�#####
	 ###�#######�#####################�####�##########################�
	 ###�#�########�###�##�#####�#################################�###
	 ##�#####�############################�#�#########################
	 ##########�#######�#�###########################�#############�##
	 ######�#�##�#��###########�#######���##�######���#######�########
	 #########�#############�###################�#####�##��##�#######���	 ###�######################��##############�########�##########�##����#########�#######�##########�##�###########################�####�
	 #########�########################�########��###�#�##############����#####�##########�############################�####�##############
	 ###�#####�#�#############�#######�#############################�#
	 ###��#####�###�###�###########�#########�###################�####
	 ############�###�######�###########�#############################
	 #######�######��#############�########�###�###########�##########
	 ####�####################�######################################�
	 ###########�################################�##�############�####
	 #######�#########�####################################�#�########
	 #�#�##�#######��########�####��##############�##########�###�###�
	 #��################�###########�##########�##########�###########
	 #�################
done
Bytes trans�����d = 11403264 (ae0000 hex)
NetBootFileXferSize�����e0000
 ������g image to 0x200000
ranand_erase_write: offs:200������ount:ae0000
............�..�.......................................................................Done!
 Writing image to ������00
ranand_erase_write: offs:600000, count:ae0000��...��....................�...................................���...���..............�..Done!
## Booting image at bc200000 ...
   Image Name:   MIPS OpenWrt Linux-4.14.131
   Data Size:    1607812 Bytes =  1.5 MB    �����mage Type:   MIPS Linux Kernel Image (lzma compres��䩊
   Load�����ess: 80000000
   Entry Point:  80000000
....��......�............   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK
commandline uart_en=1 factory_mode=0 ������8m
No initrd
## Transferring control to Linux (�����ress 80000000) ...
## Giving linux memsize in MB������
[... 215 lines cut ...]
[    0.656223] io scheduler deadline regist���䠨default)
[    0.662147] Serial: 8250/16550 driver,�����rts, IRQ sharing disabled
[   ������307] console [ttyS0] disabled
[    0.672826] 10000c00.uartli�庠��yS0 at MMIO 0x10000c00 (irq = 20, base_baud = 250����� is a Palmchip BK-3103
[    0.682449] console [ttyS0] enabled
[    0.682449] console [ttyS0] enabled
[    0.689479] bootconsole [early0] disabled
[    0.689479] bootconsole [early0] disabled
[    0.698225] cacheinfo: Failed to find cpu0 device node
[    0.703512] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.710973] libphy: Fixed MDIO Bus: probed
[    0.726143] gsw: setting port4 to ephy mode
[    0.730464] mtk_soc_eth 10100000.ethernet eth0 (uninitialized): port 4 link up (100Mbps/Full duplex)
[    0.739874] mtk_soc_eth 10100000.ethernet: generated random MAC address c6:0f:3d:5d:9e:c2
[    0.748326] mtk_soc_eth 10100000.ethernet: loaded mt7620 driver
[    0.755075] mtk_soc_eth 10100000.ethernet eth0: mediatek frame engine at 0xb0100000, irq 5
[    0.764063] rt2880_wdt 10000120.watchdog: Initialized
[    0.770570] NET: Registered protocol family 10
[    0.779207] Segment Routing with IPv6
[    0.783143] NET: Registered protocol family 17
[    0.787730] 8021q: 802.1Q VLAN Support v1.8
[    0.795015] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    0.802729] Please append a correct "root=" boot option; here are the available partitions:
[    0.811254] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)
[    0.820808] Rebooting in 1 seconds..

hanandjun, 2019-07-30:
烦请大佬过目，看一下是哪里出了问题

hanandjun, 2019-07-30:
感谢

qinfeng8848, 2019-07-31:
我分享一下我的方法：
1.先刷breed-mt7621-xiaomi-r3g.bin
2.然后在刷入openwrt-ramips-mt7621-mir3g-squashfs-kernel.bin
3.最后ssh登陆到openwrt 
root@OpenWrt:/tmp# mtd write openwrt-ramips-mt7621-mir3g-squashfs-kernel1.bin kernel_stock
root@OpenWrt:/tmp# mtd write openwrt-ramips-mt7621-mir3g-squashfs-kernel1.bin kernel
root@OpenWrt:/tmp# mtd write openwrt-ramips-mt7621-mir3g-squashfs-rootfs0.bin ubi

hanandjun, 2019-08-01:
@qinfeng8848 你好，试了一下还是不行，但是很感谢你的解答。
我对比了一下启动日志，发现正常的启动应该是会在
[    0.950617] 8021q: 802.1Q VLAN Support v1.8
之后有
[    0.959774] UBI: auto-attach mtd9
所以，这个可能是要在 kernel 开启什么选项，我会继续探索

hanandjun, 2019-08-01:
不过连分区都没打印出来，这是什么问题呢

hanandjun, 2019-08-01:
已刷入，剩下的再慢慢探索，感谢

closed 2019-08-01 by hanandjun

LiuJiangshan, 2019-08-07:
> 已刷入，剩下的再慢慢探索，感谢

能正常使用吗？

hanandjun, 2019-08-07:
@LiuJiangshan 能启动，不能正常使用😅，可以到打开的 #1523 讨论