coolsnowwolf/lede #1506: v2ray的tls只能用allowInsecure

YLaido, 2019-07-31:
x86平台，固件版本：OpenWrt R9.7.6 / LuCI Master (git-19.146.62144-fd6fdb2)，证书是有效的let's encrypt，本地v2rayN没有任何问题，路由不勾选allowInsecure就出不去，怀疑是证书问题，但是ca-bundle和ca-certificates 都已更新至最新：
```
root@OpenWrt:~# opkg list-installed |grep ca-
ca-bundle - 20190110-1
ca-certificates - 20190110-1
```
实在没办法了，希望作者能看一下。
感谢无私付出。

riverscn, 2019-07-31:
po一下你的web服务器证书用的什么。
如果你用的是Let's Encrypt，记得要指定 fullchain的key。我曾经遇到你一样的问题就是因为这个。

YLaido, 2019-07-31:
> po一下你的web服务器证书用的什么。
> 如果你用的是Let's Encrypt，记得要指定 fullchain的key。我曾经遇到你一样的问题就是因为这个。

用的确实是let，但是tls是直接nginx负责的，v2ray配置里面只有websocket，看起来这样的v2配置里不能指定证书吧

riverscn, 2019-08-01:
nginx的配置里看看是不是指定的 fullchain.cer 而不是 域名.cer。我之前就是这个问题。

YLaido, 2019-08-02:
感谢建议，但是确实用的是域名.cer

riverscn, 2019-08-02:
额，我说得很清楚了啊，你不能用域名.cer，你得用fullchain.cer。你的问题就出在这儿。

YLaido, 2019-08-03:
> 额，我说得很清楚了啊，你不能用域名.cer，你得用fullchain.cer。你的问题就出在这儿

原来如此，替换成了fullchain之后问题确实解决了。
十分感谢 :>

closed 2019-08-03 by YLaido