coolsnowwolf/lede #1513: 内核

Heartbeats1104, 2019-08-01:


adamw92, 2019-08-02:
> 
编译切换内核：
打开LEDE目录/target/linux/架构/Makefile
KERNEL_PATCHVER:=4.9
具体支持内核版本
include/kernel-version.mk
>

Heartbeats1104, 2019-08-02:
我问的是怎么更换到3.14内核，别看见你会一点点的就乱答，OK?  相应架构Makefile里的全是4.x的内核，这谁都知道。更换为4.X内核方法，Google一大堆，谁还会来这里问？

coolsnowwolf, 2019-08-02:
没那么简单，要自己写内核 patch backport 回去的

adamw92, 2019-08-02:
> 我问的是怎么更换到3.14内核，别看见你会一点点的就乱答，OK? 相应架构Makefile里的全是4.x的内核，这谁都知道。更换为4.X内核方法，Google一大堆，谁还会来这里问？

友好度满分

MeIsReallyBa, 2019-08-04:
就你这态度谁想回答

Heartbeats1104, 2019-08-04:
> 就你这态度谁想回答

哦豁？我这态度？我这态度怎么了？我请教大雕，对大雕用的是您，用的是请问等词。我问过你？还是问了哪个其他人。我叫adamw92这位兄弟别不看问题，答非所问，我态度就有问题？你怕不是傻子吧？

closed 2019-08-04 by coolsnowwolf

vft9nesd, 2019-08-05:
> > 就你这态度谁想回答
> 
> 哦豁？我这态度？我这态度怎么了？我请教大雕，对大雕用的是您，用的是请问等词。我问过你？还是问了哪个其他人。我叫adamw92这位兄弟别不看问题，答非所问，我态度就有问题？你怕不是傻子吧？

硬核选手

24kcsplus, 2023-08-12:
> > 就你这态度谁想回答
> 
> 哦豁？我这态度？我这态度怎么了？我请教大雕，对大雕用的是您，用的是请问等词。我问过你？还是问了哪个其他人。我叫adamw92这位兄弟别不看问题，答非所问，我态度就有问题？你怕不是傻子吧？

逆天