coolsnowwolf/lede #1514: 无法自动挂载exFAT

YanminZhou, 2019-08-03:
自动挂载可以挂载FAT32, 但不能自动挂载exFAT。插入exFAT的U盘后挂载点页面无变化。
编译时已选中 Auto Mount
使用`mount -t exfat /dev/sda /mnt` 命令可手动挂载，但挂载后无法卸载

机型 WR1200JS

hianghokung, 2019-08-08:
Kernel modules > Filesystems > kmod-fs-exfat
　　这一项是默认选中的，按理没问题，建议重新格式化U盘/更换U盘进行测试。

YanminZhou, 2019-08-16:
> ```
>     Kernel modules > Filesystems > kmod-fs-exfat
> ```
> 
> 　　这一项是默认选中的，按理没问题，建议重新格式化U盘/更换U盘进行测试。

 kmod-fs-exfat 已经编译在系统内；更换U盘或重新格式化U盘无效。
插入U后出现了这些日志：
`Fri Aug 16 11:46:29 2019 kern.info kernel: [62746.895451] usb 1-2: new high-speed USB device number 3 using xhci-mtk
Fri Aug 16 11:46:29 2019 kern.info kernel: [62747.077831] usb-storage 1-2:1.0: USB Mass Storage device detected
Fri Aug 16 11:46:29 2019 kern.info kernel: [62747.087218] scsi host0: usb-storage 1-2:1.0
Fri Aug 16 11:46:30 2019 kern.notice kernel: [62748.146212] scsi 0:0:0:0: Direct-Access     Netac    OnlyDisk         PMAP PQ: 0 ANSI: 6
Fri Aug 16 11:46:30 2019 kern.notice kernel: [62748.157966] sd 0:0:0:0: [sda] 15466496 512-byte logical blocks: (7.92 GB/7.38 GiB)
Fri Aug 16 11:46:30 2019 kern.notice kernel: [62748.167069] sd 0:0:0:0: [sda] Write Protect is off
Fri Aug 16 11:46:30 2019 kern.debug kernel: [62748.171858] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
Fri Aug 16 11:46:30 2019 kern.notice kernel: [62748.173225] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
Fri Aug 16 11:46:30 2019 kern.notice kernel: [62748.283548] sd 0:0:0:0: [sda] Attached SCSI removable disk`

lqzhgood, 2019-09-07:
我也是。
日志显示
```
[   16.370444] usb usb2-port1: attempt power cycle
[   16.936455] usb 2-1: new full-speed USB device number 4 using ohci-platform
[   17.380215] usb 2-1: device not accepting address 4, error -62
[   17.590220] usb 2-1: new full-speed USB device number 5 using ohci-platform
[   18.030183] usb 2-1: device not accepting address 5, error -62
[   18.036143] usb usb2-port1: unable to enumerate USB device
```

closed 2019-10-18 by coolsnowwolf