coolsnowwolf/lede #1519: IPSEC 链接10s左右即断线

bobcc4, 2019-08-05:
安卓手机不会断开连接，但 进不去路由器后台以及不能上网。用PC链接（不同网络）shrew vpn
链接10s及断开连接，显示gateway断开。（已咨询群友，并非个例）

主机名 | OpenWrt
-- | --
主机型号 | Intel(R) Celeron(R) CPU J1900 @ 1.99GHz
CPU信息 | 1332.718 MHz +41.0°C
固件版本 | OpenWrt R9.8.5 / LuCI Master (git-19.146.62144-fd6fdb2)
内核版本 | 4.9.186
本地时间 | Mon Aug 5 22:40:57 2019
运行时间 | 1h 26m 57s
平均负载 | 0.00, 0.00, 0.00

日志
Mon Aug  5 22:32:23 2019 daemon.info : 05[IKE] received DPD vendor ID
Mon Aug  5 22:32:23 2019 daemon.info : 05[IKE] 49.94.43.18 is initiating a Main Mode IKE_SA
Mon Aug  5 22:32:23 2019 authpriv.info : 05[IKE] 49.94.43.18 is initiating a Main Mode IKE_SA
Mon Aug  5 22:32:23 2019 daemon.info : 05[ENC] generating ID_PROT response 0 [ SA V V V V ]
Mon Aug  5 22:32:23 2019 daemon.info : 05[NET] sending packet: from 221.227.174.192[500] to 49.94.43.18[13702] (160 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 07[NET] received packet: from 49.94.43.18[13702] to 221.227.174.192[500] (228 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 07[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
Mon Aug  5 22:32:23 2019 daemon.info : 07[IKE] remote host is behind NAT
Mon Aug  5 22:32:23 2019 daemon.info : 07[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
Mon Aug  5 22:32:23 2019 daemon.info : 07[NET] sending packet: from 221.227.174.192[500] to 49.94.43.18[13702] (244 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 14[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (92 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 14[ENC] parsed ID_PROT request 0 [ ID HASH ]
Mon Aug  5 22:32:23 2019 daemon.info : 14[CFG] looking for XAuthInitPSK peer configs matching 221.227.174.192...49.94.43.18[10.43.18.48]
Mon Aug  5 22:32:23 2019 daemon.info : 14[CFG] selected peer config "xauth_psk"
Mon Aug  5 22:32:23 2019 daemon.info : 14[ENC] generating ID_PROT response 0 [ ID HASH ]
Mon Aug  5 22:32:23 2019 daemon.info : 14[NET] sending packet: from 221.227.174.192[4500] to 49.94.43.18[13703] (76 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 14[ENC] generating TRANSACTION request 3292117349 [ HASH CPRQ(X_USER X_PWD) ]
Mon Aug  5 22:32:23 2019 daemon.info : 14[NET] sending packet: from 221.227.174.192[4500] to 49.94.43.18[13703] (76 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 09[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (108 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 09[ENC] parsed INFORMATIONAL_V1 request 2324340099 [ HASH N(INITIAL_CONTACT) ]
Mon Aug  5 22:32:23 2019 daemon.info : 06[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (108 bytes)
Mon Aug  5 22:32:23 2019 daemon.info : 06[ENC] parsed TRANSACTION response 3292117349 [ HASH CPRP(X_USER X_PWD) ]
Mon Aug  5 22:32:23 2019 daemon.info : 06[IKE] XAuth authentication of 'song' successful
Mon Aug  5 22:32:23 2019 daemon.info : 06[ENC] generating TRANSACTION request 2913779380 [ HASH CPS(X_STATUS) ]
Mon Aug  5 22:32:23 2019 daemon.info : 06[NET] sending packet: from 221.227.174.192[4500] to 49.94.43.18[13703] (76 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 08[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (92 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 08[ENC] parsed TRANSACTION response 2913779380 [ HASH CPA(X_STATUS) ]
Mon Aug  5 22:32:24 2019 daemon.info : 08[IKE] IKE_SA xauth_psk[1] established between 221.227.174.192[221.227.174.192]...49.94.43.18[10.43.18.48]
Mon Aug  5 22:32:24 2019 authpriv.info : 08[IKE] IKE_SA xauth_psk[1] established between 221.227.174.192[221.227.174.192]...49.94.43.18[10.43.18.48]
Mon Aug  5 22:32:24 2019 daemon.info : 08[IKE] scheduling reauthentication in 9732s
Mon Aug  5 22:32:24 2019 daemon.info : 08[IKE] maximum IKE_SA lifetime 10272s
Mon Aug  5 22:32:24 2019 daemon.info : 11[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (124 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 11[ENC] parsed TRANSACTION request 3835872822 [ HASH CPRQ(ADDR MASK DNS NBNS U_BANNER U_DEFDOM U_SPLITDNS U_SPLITINC U_LOCALLAN VER) ]
Mon Aug  5 22:32:24 2019 daemon.info : 11[IKE] peer requested virtual IP %any
Mon Aug  5 22:32:24 2019 daemon.info : 11[CFG] assigning new lease to 'song'
Mon Aug  5 22:32:24 2019 daemon.info : 11[IKE] assigning virtual IP 192.168.1.10 to peer 'song'
Mon Aug  5 22:32:24 2019 daemon.info : 11[ENC] generating TRANSACTION response 3835872822 [ HASH CPRP(ADDR DNS) ]
Mon Aug  5 22:32:24 2019 daemon.info : 11[NET] sending packet: from 221.227.174.192[4500] to 49.94.43.18[13703] (92 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 13[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (668 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 13[ENC] parsed QUICK_MODE request 3957819306 [ HASH SA No ID ID ]
Mon Aug  5 22:32:24 2019 daemon.info : 13[IKE] received 28800s lifetime, configured 3600s
Mon Aug  5 22:32:24 2019 daemon.info : 13[ENC] generating QUICK_MODE response 3957819306 [ HASH SA No ID ID ]
Mon Aug  5 22:32:24 2019 daemon.info : 13[NET] sending packet: from 221.227.174.192[4500] to 49.94.43.18[13703] (172 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 12[NET] received packet: from 49.94.43.18[13703] to 221.227.174.192[4500] (76 bytes)
Mon Aug  5 22:32:24 2019 daemon.info : 12[ENC] parsed QUICK_MODE request 3957819306 [ HASH ]
Mon Aug  5 22:32:24 2019 daemon.info : 12[IKE] CHILD_SA xauth_psk{1} established with SPIs c8da26b4_i 025ac9b7_o and TS 0.0.0.0/0 === 192.168.1.10/32
Mon Aug  5 22:32:24 2019 authpriv.info : 12[IKE] CHILD_SA xauth_psk{1} established with SPIs c8da26b4_i 025ac9b7_o and TS 0.0.0.0/0 === 192.168.1.10/32
Mon Aug  5 22:33:26 2019 daemon.err uhttpd[2239]: luci: accepted login on / for root from 192.168.1.184

ilasx, 2019-08-07:
我的android和ios连上都只能访问路由器，不能打开网页

bobcc4, 2019-08-26:
问题已解决，手动修改内核为4.14，重新编译即可，4.9内核官方bug

ilasx, 2019-09-15:
> 问题已解决，手动修改内核为4.14，重新编译即可，4.9内核官方bug

大哥，我修改了还是不行，能编译一个给我用吗？

closed 2019-10-18 by coolsnowwolf