coolsnowwolf/lede #1523: MIWIFI ROUTER R3 cat /proc/mtd 为空

hanandjun, 2019-08-06:
刷入启动后的日志：

Byte?????sferred = 6240445 (5f38bd hex)
??????tFileXferSize= 005f38bd
 Writing ?????to 0x200000
ranand_erase_write: offs:200000, cou?????8bd
....???.....?................................?...Done!
 Writing ?????to 0x600000
ranand_erase_write: offs:600000, cou?????8bd
....?.....???.....................?..............Done!
## Booting image at bc200000 ...
?  Image Name:   MIPS OpenWrt Linux-4.14.134
   Im????ype:   MIPS Linux Kernel Image (lzma compressed)
?????ta Size:    6240381 Bytes =  6 MB
   Load Addres󺠸??00000
   Entry Point:  80000000
...?.................?......?.......?................??.....??......?.........?.......?......?..???Verifying Checksum ... Bad Da?????

U-Boot 1.1.3 (Apr 15 2016 - 17:46:32)

Boar亠???ink APSoC DRAM:  128 MB
Power on memory test. Me??????ize= 128 MB...OK!
relocate_code Pointer at: 87fb8000
enable ephy clock...done. rf re砲??= 5
SSC disabled.
******************************?????tware System Reset Occurred
*********************?????****
!!! nand page size = 2048, addr len=4
..???========================================= 
Ralin?????t Version: 4.2.S.1
-----------------------------??????--------- 
ASIC 7620_MP (Port5<->None)
DRAM_CON??????: Auto-detection 
DRAM_TYPE: DDR2 
DRAM compone??????24 Mbits
DRAM bus: 16 bit
Total memory: 128 MBy???Flash component: NAND Flash
Date:Apr 15 2016  Tim庱??46:32
===========================================?????cache: sets:512, ways:4, linesz:32 ,total:65536
d?????: sets:256, ways:4, linesz:32 ,total:32768 

 ##?????he CPU freq = 580 MHZ #### 
 estimate memory size????? Mbytes

Please?????se the operation: 
   1: Load system code to SDRA͠??? TFTP. 
   2: Load system code then write to Flas???? TFTP. 
   3: Boot system code via Flash (default?????  4: Entr boot command line interface.
   9: Load????? Loader code then write to Flash via TFTP. 
 4 1 ??????
   
3: System Boot system code via Flash.
Boot failure detected on both systems
Ver?????? kernel1 uImage CRC, addr: 0xbc200000
   Image Name:   MIPS OpenWrt Linux-4??????4
   Image Type:   MIPS Linux Kernel Image (lzma??????essed)
   Data Size:    6240381 Bytes =  6 MB
 ?????? Address: 80000000
   Entry Point:  80000000
.......................?..?..?..?..............................?..?..?..........................   Verifying ?????sum ... Bad Data CRC
Verifyi?????nel2 uImage CRC, addr: 0xbc600000
?????age Name:   MIPS OpenWrt Linux-4.14.134
   Image?????   MIPS Linux Kernel Image (lzma compressed)
   ?????ize:    6240381 Bytes =  6 MB
   Load Address: 8??????0
   Entry Point:  80000000
..................?..??..?..?.............................??..?..?.............................?   V?????ing Checksum ... OK
Booting System 2
..ranand_er??????ite: offs:40000, count:20000
?????
[... 187 lines cut ...]
[   11.615752] random: jshn: uninitialized urandom read (4 bytes read)
[   11.699748] random: jshn: uninitialized urandom read (4 bytes read)
[   18.135751] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   18.135751] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   26.932115] 8021q: adding VLAN 0 to HW filter on device eth0
[   26.944189] br-lan: port 1(eth0.1) entered blocking state
[   26.949793] br-lan: port 1(eth0.1) entered disabled state
[   26.955654] device eth0.1 entered promiscuous mode
[   26.960727] device eth0 entered promiscuous mode
[   27.001141] br-lan: port 1(eth0.1) entered blocking state
[   27.006668] br-lan: port 1(eth0.1) entered forwarding state
[   27.012602] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   27.929258] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   30.867991] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   30.893048] br-lan: port 2(wlan0) entered blocking state
[   30.898486] br-lan: port 2(wlan0) entered disabled state

然后 
```
root@OpenWrt:/# cat /proc/mtd
dev:    size   erasesize  name
root@OpenWrt:/# 
```

烦请大佬分析下问题，感谢

hanandjun, 2019-08-06:
可能还是 kernel 配的不对🤣

LiuJiangshan, 2019-08-07:
> 可能还是 kernel 配的不对🤣
似乎在合并lede主分支代码时将mi3的适配删掉了部分
mi3适配相关提交:https://github.com/coolsnowwolf/lede/commit/42d3498f558e64f73e50609df5d72197acc90119 https://github.com/coolsnowwolf/lede/commit/ec8d2878e4e344065d24947f4eb209fb923cf431

hanandjun, 2019-08-08:
@LiuJiangshan 大兄弟，折腾了下，现在能创建分区了，但是ubi貌似挂了，所以还是不能正常使用🤣
[image]

```
Please choose the operation: 
   1: Load system ??????SDRAM via TFTP. 
   2: Load system code then write to Flash via ???Ю???   3: Boot system code via Flash (default).
   4: Entr boot comma?????? interface.
   9: Load Boot Loader code then write to Flash via T??Ю 2 ??? 0 
   
3: Sys???????? system code via Flash.
Boot failure detect??????th systems
Verifying kernel1 uImage CRC, addr: 0xbc200000
   Image Name:   MIPS OpenWrt Linux-4.14.136
   Image????庠? MIPS Linux Kernel Image (lzma compressed)
   Data Size:    6379????????s =  6.1 MB
   Load Address: 80000000
   Entry Point:  80000000
                         ???............?..?..?...........?..?..?..............?..?..............?..?..............?........   Verifying Checksum ... ???Data CRC
Verifying kernel2 uImage CRC, addr: 0xbc600000
   Image Name:   MIPS OpenWrt Linux-4.????????   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Da??????    6379130 Bytes =  6.1 MB
   Load Address: 80000000
   Entry ????????80000000
............?..?..?...........?..?..?...........?..?...........?..?..?...........?..?...........?.   Verifying Checksum ... OK
Booting System 2
?.ranand_erase_write: offs:40000, count:20000
.Done!
done
## Booting image at bc600000 ...
   Image Name:   MIPS ????????Linux-4.14.136
   Image Type:   MIPS Linux Kernel Image (lzma co????????)
   Data Size:    6379130 Bytes =  6.1 MB
   Load Address: 800????????  Entry Point:  80000000
..........?....?................?....??...............??...??...............?....?................   Verifying Checksum ... OK
   Uncompressing Ker???????e ... OK
commandline uart_en=1 factory_mode=0 mem=128m
No initrd
#????????erring control to Linux (at address 80000000) ...
## Giving linux???????e in MB, 128

Starting kernel ...

[    0.000000] Linux version 4.14.136 (hansai@ha????????cBook-Air) (gcc version 7.4.0 (OpenWrt GCC 7.4.0 r1433-7f39235e))???????Aug 6 13:10:35 2019
[    0.000000] Board has DDR2
[    0.000000ݠ?????? PMU set to hw control
[    0.000000] Digital PMU set to hw contr?썊۠?  0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
[    0.000000]????????sole [early0] enabled
[    0.000000] CPU0 revision is: 00019650 ????Ӡ??KEc)
[    0.000000] MIPS: machine is???????Mi Router R3
[    0.000000] Determined???????al RAM map:
?[    0.000000] Initrd not found or empty????????ling initrd
[    0.000000] Primary instruction cache 64kB, VIPT,????????linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way,????Ԭ?no aliases, linesize 32 bytes
[    0.000000] Zone r????󺍊[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007fff???ݍ?۠   0.000000] Movable zone start for each node
[    0.000000] Ear????????y node ranges
[    0.000000]   node   0: [mem 0x0000000000000000????????00007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x000000000???????-0x0000000007ffffff]
[    0.000000] random: get_ra????????es called from start_kernel+0x90/0x478 with crng_init=0
[    0.000000] Built 1 z????????, mobility grouping on.  Total pages: 32512
[    0.000000] Kerne???????d line: console=ttyS0,115200 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order????????48 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (orde򺠴????536 bytes)
[    0.000000] In???????e hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00037d90
[   ????????0] Readback ErrCtl register=00037d90
[    0.0?????ݠ?emory: 118724K/131072K available (4081K kernel code, 178K rwdata,????ˠ??data, 5680K init, 213K bss, 12348K reserved, 0K cma-reserved)
۠?????000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=???[    0.0?????ݠ?R_IRQS: 256
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocks????庠?ystick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns??۠???0.000000] systick: enable autosleep mode
[... 169 lines cut ...]
[   36.231888] ieee80211 phy1: rt2800_iq_search: Info - IQCalibration Done! CH = 1, (gain= 0, phase=3a)
[   36.241249] ieee80211 phy1: rt2800_loft_iq_calibration: Info - TX IQ Calibration Done!
[   36.439610] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=2, Ph_rx=-4
[   36.457602] ieee80211 phy1: rt2800_rxiq_calibration: Info - RXIQ G_imb=0, Ph_rx=-3
[   36.516687] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[   36.593358] br-lan: port 2(wlan1) entered blocking state
[   36.598894] br-lan: port 2(wlan1) entered disabled state
[   36.604748] device wlan1 entered promiscuous mode
[   37.091625] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   37.141383] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
[   37.148065] br-lan: port 2(wlan1) entered blocking state
[   37.153566] br-lan: port 2(wlan1) entered forwarding state
[   37.164080] br-lan: port 3(wlan0) entered blocking state
[   37.169745] br-lan: port 3(wlan0) entered disabled state
[   37.175529] device wlan0 entered promiscuous mode
[   37.180522] br-lan: port 3(wlan0) entered blocking state
[   37.185948] br-lan: port 3(wlan0) entered forwarding state
[   37.539337] br-lan: port 3(wlan0) entered disabled state
[   38.009154] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   38.015867] br-lan: port 3(wlan0) entered blocking state
[   38.021376] br-lan: port 3(wlan0) entered forwarding state

```

其中 UBI error: no valid UBI magic found inside mtd9 这个应该是提示信息

hanandjun, 2019-08-11:
@LiuJiangshan 老哥，解决了，应该是缺了个补丁，还有kernel配置下应该就可以了，我那个错看了openwrt 一个设备的wiki说是坏块的什么问题，反正后来格式化之后重新attach就好了😆可以愉快的玩耍了

[image]
[image]

closed 2019-08-11 by hanandjun

LiuJiangshan, 2019-08-11:
@hanandjun 能把修改的代码提个push request吗？

hanandjun, 2019-08-13:
@LiuJiangshan 大兄弟，待会儿我编译一个发给你，你刷一下试试，先不提pl了，我改动的地方还不是很明白什么意思，就不污染代码了

LiuJiangshan, 2019-08-13:
@hanandjun fork出来提交，或则代码打包发到我的邮箱ljs234@qq.com

dingpengyu, 2019-12-08:
小米3g出现了同样的问题，说真的挺着急的。

hanandjun, 2019-12-08:
> 小米3g出现了同样的问题，说真的挺着急的。

看我fork的仓库，里面第一条提交