coolsnowwolf/lede #1525: SSR+是否优先端口转发

alexyw, 2019-08-07:
K3，没用V2ray，只SS（不带R），遇到的情况如下，不知如何处理
只要打开SS，开始端口转发还能用，但基本不出半天，端口转发就无效了，路由器温度正常，WIFI功能正常，上网也正常，就转发失效了，换其他端口都是一样的，我特意每3000个号段挑一个端口的。
SS一切默认值，没开游戏加速。
编译时config如下：
`CONFIG_TARGET_bcm53xx=y
CONFIG_TARGET_bcm53xx_DEVICE_phicomm-k3=y
CONFIG_TARGET_BOARD="bcm53xx"
CONFIG_LIBSODIUM_MINIMAL=y
# CONFIG_PACKAGE_kmod-crypto-aead is not set
# CONFIG_PACKAGE_kmod-crypto-ecb is not set
# CONFIG_PACKAGE_kmod-crypto-hash is not set
# CONFIG_PACKAGE_kmod-crypto-manager is not set
# CONFIG_PACKAGE_kmod-crypto-null is not set
# CONFIG_PACKAGE_kmod-crypto-pcompress is not set
# CONFIG_PACKAGE_kmod-crypto-sha1 is not set
# CONFIG_PACKAGE_kmod-gre is not set
# CONFIG_PACKAGE_kmod-mppe is not set
# CONFIG_PACKAGE_kmod-nf-conntrack-netlink is not set
# CONFIG_PACKAGE_kmod-tun is not set
# CONFIG_PACKAGE_kmod-usb-printer is not set
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libev=y
CONFIG_PACKAGE_libmbedtls=y
# CONFIG_PACKAGE_libminiupnpc is not set
# CONFIG_PACKAGE_libnatpmp is not set
CONFIG_PACKAGE_libsodium=y
# CONFIG_PACKAGE_libstdcpp is not set
# CONFIG_PACKAGE_luci-app-nlbwmon is not set
# CONFIG_PACKAGE_luci-app-pptp-server is not set
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray=y
# CONFIG_PACKAGE_luci-app-usb-printer is not set
# CONFIG_PACKAGE_luci-app-vsftpd is not set
# CONFIG_PACKAGE_luci-app-wifischedule is not set
# CONFIG_PACKAGE_luci-app-xlnetacc is not set
# CONFIG_PACKAGE_luci-app-zerotier is not set
# CONFIG_PACKAGE_nlbwmon is not set
# CONFIG_PACKAGE_p910nd is not set
# CONFIG_PACKAGE_pptpd is not set
CONFIG_PACKAGE_shadowsocks-libev-config=y
CONFIG_PACKAGE_shadowsocks-libev-ss-redir=y
CONFIG_PACKAGE_v2ray=y
# CONFIG_PACKAGE_vsftpd-alt is not set
# CONFIG_PACKAGE_wifischedule is not set
# CONFIG_PACKAGE_zerotier is not set
`

mrclock, 2019-08-08:
V2也有这个问题。感觉SSR+会优先(劫持)数据，不会进入端口转发。不过你用绕开国内IP试试看。端口转发就生效了。我的情况是，"劫持"的数据全部到对应的V2服务器上去了，吓得我现在都不敢开。不知道你的是不是这个情况，你可以上你SS服务器看下log

alexyw, 2019-08-08:
国内IP模式会导致国内访问变慢，毕竟都是谷歌DNS解析的地址。
想搞一个更彻底的办法，不然很烦
我的ss看不了日志

mrclock, 2019-08-08:
> 国内IP模式会导致国内访问变慢，毕竟都是谷歌DNS解析的地址。
> 想搞一个更彻底的办法，不然很烦
> 我的ss看不了日志

绕开国内IP模式，国内访问不会变慢，因为被劫持的基本都是国内IP。只有这样，才会放过这些流量，进入端口转发。只是临时解决方案。用PT这种 可能就会大量用SS流量了。

我看了。用绕开国内IP模式，v2服务器上基本没特殊日志。

coolsnowwolf, 2019-08-08:
并不会，测试过不会发生这个现象

mrclock, 2019-08-08:
> 并不会，测试过不会发生这个现象

大雕，有时候并不会立马出现，可能10分钟，可能1分钟。真说不准。也不知道哪里有问题。原先用的4.9内核，刚编译了个4.19 晚上回去再试试看。

mrclock, 2019-08-09:
X86的，用最新的源码编译了一个4.19内核，
只添加了zerotier 和 unblockmusic。

刚开始 运行SSR+没有发生问题，但是10分钟后，就抽风了。入站数据又被拦截转发到v2服务器上去了。

zerotier启用，并打开了【自动允许客户端NAT】

DnsMasq 增加了以下内容
#4KIPTV需要
dhcp-option-force=125,00:00:00:00:1a:02:06:48:47:57:2d:43:54:03:08:45:2d:31:34:30:57:2d:50:0b:02:00:55:0a:02:20:00:3c:1e:00:00:01:00:02:03:43:50:45:03:0e:45:38:20:45:50:4f:4e:20:52:4f:55:54:45:52:04:03:31:2e:30
dhcp-option=15
dhcp-option=28
dhcp-option=60,00:00:01:06:68:75:61:71:69:6E:02:0A:48:47:55:34:32:31:4E:20:76:33:03:0A:48:47:55:34:32:31:4E:20:76:33:04:10:32:30:30:2E:55:59:59:2E:30:2E:41:2E:30:2E:53:48:05:04:00:01:00:50

并0,1,2口做了个Vlan85

设置了一点端口转发，其他没啥特殊配置。

unblockmusic我觉得问题不大，因为以前没这货也有这个情况。

alexyw, 2019-08-09:
https://github.com/coolsnowwolf/lede/issues/1525#issuecomment-519464811

那是因为你测试时间不够长，我开SS基本上6个小时内不出现这情况，半天就说不准了，反正上周末出去钓鱼，就连不上家里的任何设备了。

国内IP是国外DNS解析国内的域名，与国内CDN肯定又不一致的，虽然影响不大，但确实不舒服。

周一起我关掉了SSR，至目前一切转发正常。但发现相对官方18.06的转发还是有一点小问题，就是远程桌面时明显慢一些，官方的秒开

alexyw, 2019-08-14:
这个问题是否无解了？
可怜我的K3

hyyz17200, 2019-08-14:
> 
> 
> [#1525 (comment)](https://github.com/coolsnowwolf/lede/issues/1525#issuecomment-519464811)
> 
> 那是因为你测试时间不够长，我开SS基本上6个小时内不出现这情况，半天就说不准了，反正上周末出去钓鱼，就连不上家里的任何设备了。
> 
> 国内IP是国外DNS解析国内的域名，与国内CDN肯定又不一致的，虽然影响不大，但确实不舒服。
> 
> 周一起我关掉了SSR，至目前一切转发正常。但发现相对官方18.06的转发还是有一点小问题，就是远程桌面时明显慢一些，官方的秒开

远程桌面慢和FlowOffload有关，关闭就可以了

hyyz17200, 2019-08-14:
> X86的，用最新的源码编译了一个4.19内核，
> 只添加了zerotier 和 unblockmusic。
>
> 刚开始 运行SSR+没有发生问题，但是10分钟后，就抽风了。入站数据又被拦截转发到v2服务器上去了。
>
> zerotier启用，并打开了【自动允许客户端NAT】
>
> DnsMasq 增加了以下内容
> #4KIPTV需要 [...]

刚想到是不是你也开了Flow，把Flow关了再看看？Flow开了转发的问题很多。

alexyw, 2019-08-15:
远程桌面还真是flowoffload在作怪，关了后速度猛升！谢谢
不过我SSr已经删除了，懒得去测试了，太酸爽

hyyz17200, 2019-08-15:
> 
> 
> 远程桌面还真是flowoffload在作怪，关了后速度猛升！谢谢
> 不过我SSr已经删除了，懒得去测试了，太酸爽

嗯没错。
我昨晚刚更新了最新编译，网易云破解 V2RAYGFW模式 远程桌面转发 三个全开，短时间内看没有问题。希望mrclock老兄也是这个Flow问题。

mrclock, 2019-08-15:
> > 远程桌面还真是flowoffload在作怪，关了后速度猛升！谢谢
> > 不过我SSr已经删除了，懒得去测试了，太酸爽
> 
> 嗯没错。
> 我昨晚刚更新了最新编译，网易云破解 V2RAYGFW模式 远程桌面转发 三个全开，短时间内看没有问题。希望mrclock老兄也是这个Flow问题。

我刚试了，关了Flow Offloading，一打开V2ray，远程v2服务器上立马一堆日志。
2019/08/15 11:10:20 tcp:114.94.192.79:53872 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:20 tcp:114.94.192.79:53156 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:20 tcp:114.94.192.79:50874 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:20 tcp:114.94.192.79:54852 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:20 tcp:114.94.192.79:51450 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:20 tcp:114.94.192.79:50422 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:54070 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:54024 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:50974 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:54128 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:53024 accepted tcp:114.94.192.79:1433
2019/08/15 11:10:21 tcp:114.94.192.79:50226 accepted tcp:114.94.192.79:1433

都是别人扫1433端口的，实质上我转发根本就没设置过1433。

mrclock, 2019-08-15:
> 刚想到是不是你也开了Flow，把Flow关了再看看？Flow开了转发的问题很多。

还是不行，哎。基本放弃了。

coolsnowwolf, 2019-08-16:
已修复使用 v2ray 的时候，大量连接被转发到服务器上的问题。 
GFW模式的和 V2RAY 之间特有的问题。

mrclock, 2019-08-19:
> 已修复使用 v2ray 的时候，大量连接被转发到服务器上的问题。
> GFW模式的和 V2RAY 之间特有的问题。

大雕，刚编译了最新版，保留配置刷了下，一打开V2，又爆了。。。 

-_-|||

alexyw, 2019-08-19:
> 
> 
> > 已修复使用 v2ray 的时候，大量连接被转发到服务器上的问题。
> > GFW模式的和 V2RAY 之间特有的问题。
> 
> 大雕，刚编译了最新版，保留配置刷了下，一打开V2，又爆了。。。
> 
> -_-|||

你可以清空配置重新测试下么：）
可惜我已经放弃，连K3都卖掉了

mrclock, 2019-08-20:
> > > 已修复使用 v2ray 的时候，大量连接被转发到服务器上的问题。
> > > GFW模式的和 V2RAY 之间特有的问题。
> > 
> > 
> > 大雕，刚编译了最新版，保留配置刷了下，一打开V2，又爆了。。。
> > -_-|||
> 
> 你可以清空配置重新测试下么：）
> 可惜我已经放弃，连K3都卖掉了

我是X86的。远程刷的，清了配置就起不来了。等周末有空了试试看吧。

WROIATE, 2019-09-11:
出现同样问题在ssr上，全局模式端口转发失败，通过绕过国内正常，gfw正常，我试了下把绕过国内的ip改成我教育网ip发现也和全局模式一样端口转发失败（之前在pandora上这么做是正常的），盲猜是有特定ip必须绕过才能转发成功

closed 2019-10-18 by coolsnowwolf

zidane114, 2020-02-07:
上个月拉最新版本编译的，问题一样啊，国内ip模式就行，gfw模式端口转发就失效，基本一开就立马失效