coolsnowwolf/lede #1530: 編譯MIPS RT-N16 會出現maxlen exceeds default maximum!  Beware of overwriting nvram!

kai456789, 2019-08-07:
WARNING: maxlen exceeds default maximum!  Beware of overwriting nvram!
fread failure or file "/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/fs_mark" too large
Makefile:1002: recipe for target '/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/tmp/openwrt-brcm47xx-mips74k-asus-rt-ac53u-squashfs.trx' failed
make[5]: *** [/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/tmp/openwrt-brcm47xx-mips74k-asus-rt-ac53u-squashfs.trx] Error 1

請問我要編譯的是MIPS RT-N16的機型 但他會出現此訊息卡在NVRAM超過限制

rt-ac53u這邊先卡住 導致後面編譯不出來 是要刪除多餘套件嗎

kai456789, 2019-08-18:
> 
> 
> WARNING: maxlen exceeds default maximum! Beware of overwriting nvram!
> fread failure or file "/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/fs_mark" too large
> Makefile:1002: recipe for target '/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/tmp/openwrt-brcm47xx-mips74k-asus-rt-ac53u-squashfs.trx' failed
> make[5]: *** [/home/kai/lede/build_dir/target-mipsel_74kc_musl/linux-brcm47xx_mips74k/tmp/openwrt-brcm47xx-mips74k-asus-rt-ac53u-squashfs.trx] Error 1
> 
> 請問我要編譯的是MIPS RT-N16的機型 但他會出現此訊息卡在NVRAM超過限制
> 
> rt-ac53u這邊先卡住 導致後面編譯不出來 是要刪除多餘套件嗎
> 
> 另外能弄個LUCI 支援USB外接硬碟的套件嗎

已自行解決 請關閉

closed 2019-08-18 by kai456789

cyber-prog0x, 2020-04-09:
我也遇到类似的问题，请问您是如何解决的？