coolsnowwolf/lede #1538: ARIA2 LUCI啟動失敗

kai456789, 2019-08-09:
ARIA2 使用LUCI的啟動就會失敗 會出現一個選項或多個選項不能為空的提示

有檢查過該填的選項都填了
 
後來用SSH連入 寫好配置CONF 下命令啟動就成功

shikkoku, 2019-08-09:
确认一下设置页面的配置路径？

kai456789, 2019-08-09:
> 
> 
> 確認一下設置頁面的配置路徑？
> 例如luci配置頁面下的Download directory，你需要手動指定一個下載文件夾。默認那個路徑不存在的。
下載目錄 已指定 /mnt/sda2
系統日誌顯示 daemon.err aria2: Please create downlod dir first.

shikkoku, 2019-08-09:
> > 確認一下設置頁面的配置路徑？
> > 例如luci配置頁面下的Download directory，你需要手動指定一個下載文件夾。默認那個路徑不存在的。
> > 下載目錄 已指定 /mnt/sda2
> > 系統日誌顯示
> > Fri Aug  9 23:10:03 2019 daemon.err aria2: Please create downlod dir first.
> > Fri Aug  9 23:14:31 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
> > Fri Aug  9 23:14:31 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda2/
> > Fri Aug  9 23:14:34 2019 daemon.info aria2: Aria2 will run with user 'aria2'.
> > Fri Aug  9 23:14:34 2019 daemon.info aria2: Please make sure user 'aria2' has write access to download dir: /mnt/sda2/

提示你的文件夹没有访问权限，把sda2文件夹设置成0777权限。

kai456789, 2019-08-09:
> 提示你的文件夹没有访问权限，把sda2文件夹设置成0777权限。

感謝用此方法已解決

closed 2019-08-14 by coolsnowwolf