coolsnowwolf/lede #1541: RTL8812AU驱动不工作

FattyboyN, 2019-08-12:
树莓派4B，编译时选了RTL8812au-ct驱动。 编译完成，运行，插入usb网卡（RTL8812au），无法识别。lsusb下看有，但是进入openwrt 用ifconfig查看只有板载默认的wlan0， 没有usb网卡

hcym, 2019-08-12:
兼容性问题，pc上也是经常断线，还是ax88179稳妥

1021683053, 2019-08-13:
@hcwhan 之前我也是，但是可以手动编译，手动挂载，应该可以的，就是太麻烦了。
[https://github.com/abperiasamy/rtl8812AU_8821AU_linux](https://github.com/abperiasamy/rtl8812AU_8821AU_linux)

FattyboyN, 2019-08-13:
> @hcwhan 之前我也是，但是可以手动编译，手动挂载，应该可以的，就是太麻烦了。
> https://github.com/abperiasamy/rtl8812AU_8821AU_linux

Lean的源码里包含了一个8812au-ct的驱动，看makefile里这个驱动是从abperiasamy的repo里分枝出来的。我是编译了Lean源码里这个8812au-ct的驱动到kernel modules里。插上去后dmesg显示如下：
1.090942] usb 1-1.3: new high-speed USB device number 3 using xhci_hcd
[    1.226955] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    1.231445] usb 1-1.3: New USB device found, idVendor=0bda, idProduct=b812, bcdDevice= 2.10
[    1.231472] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    1.231490] usb 1-1.3: Product: USB3.0 802.11ac 1200M Adapter
[    1.231506] usb 1-1.3: Manufacturer: Realtek
[    1.231519] usb 1-1.3: SerialNumber: 123456
.....
5.944326] RTL871X: module init start
[    5.944342] RTL871X: rtl8812au v4.3.14_13455.20150212_BTCOEX20150128-51
[    5.944348] RTL871X: rtl8812au BT-Coex version = BTCOEX20150128-51
[    5.944420] usbcore: registered new interface driver rtl8812au
[    5.944427] RTL871X: module init ret=0

这个序列号感觉怪怪的 还有这个RTL871X不知道什么意思

用lsusb -t查看
root@OpenWrt:~# lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 3, If 0, Class=Vendor Specific Class, Driver=, 480M
        |__ Port 4: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 4: Dev 4, If 1, Class=Human Interface Device, Driver=usbhid, 12M

是不是驱动没有挂载上？

如何手动编译和挂载？

1021683053, 2019-08-13:
@FattyboyN 我也不懂呀，我很早以前再树莓派上编译过此驱动，仓库文档有怎么编译的呀。
你可以试一下。用的很稳定都是热插拔的。

FattyboyN, 2019-08-13:
> @FattyboyN 我也不懂呀，我很早以前再树莓派上编译过此驱动，仓库文档有怎么编译的呀。
> 你可以试一下。用的很稳定都是热插拔的。

是不是直接把https://github.com/abperiasamy/rtl8812AU_8821AU_linux
这里面的Makefile考到package/8812au/ 目录下，然后就行了？ 

还是说要先把这个git clone下来到本地，单独编译这个驱动？

FattyboyN, 2019-08-15:
更新：我的usb无线网卡其实是rtl8812bu，不是au，我联系了greenben，他说8812au的驱动他也有问题。

然后我在树莓派的raspbian下编译了8812bu的驱动，完全正常工作。

现在的问题是8812bu的驱动在openwrt的sdk下编译不通过。

hcym, 2019-09-24:
刚刚买个羊垃圾网件a6000就是这个，win下面用的还行，ubu就麻烦了，开机启动能正常用，不能断开，否则就再也回不去了，只能再开机，高通还没有usb的无线，活见鬼

closed 2019-10-18 by coolsnowwolf