coolsnowwolf/lede #1545: K2P 的 WIFI 设备连上之后获取不到 IP

zmz125000, 2019-08-15:
驱动用的 2.4ghz 5ghz 二选一的那个
```
Thu Aug 15 10:05:29 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 某MAC
Thu Aug 15 10:06:05 2019 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 某MAC
Thu Aug 15 10:06:06 2019 daemon.info hostapd: wlan0: STA 某MAC IEEE 802.11: authenticated
Thu Aug 15 10:06:06 2019 daemon.info hostapd: wlan0: STA 某MAC IEEE 802.11: associated (aid 1)
Thu Aug 15 10:06:06 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 某MAC
```

GHSY123, 2019-08-15:
无线驱动是哪的？是大雕里面集成的那个mt7615还是闭源的5.0.2.0驱动？

zmz125000, 2019-08-15:
@GHSY123
这个
ramips: add kmod-mt7615e to Phicomm K2P images

Now that the mt76 driver supports the MT7615D chip found on these
devices, use it.

Also add the wpad-basic package.

Note: the driver supports operation on both the 2.4 GHz and the
      5 GHz bands, but not yet concurrently.

GHSY123, 2019-08-15:
> @GHSY123
> 这个
> ramips: add kmod-mt7615e to Phicomm K2P images
> 
> Now that the mt76 driver supports the MT7615D chip found on these
> devices, use it.
> 
> Also add the wpad-basic package.
> 
> Note: the driver supports operation on both the 2.4 GHz and the
> 5 GHz bands, but not yet concurrently.

有没有链接网址什么的？

zmz125000, 2019-08-15:
@GHSY123
问一下用 5.0.2.0 驱动的话可以开启802.11r或802.11s吗？

GHSY123, 2019-08-15:
> @GHSY123
> 问一下用 5.0.2.0 驱动的话可以开启802.11r或802.11s吗？

不能，5.0.2.0驱动已经很久没更新了，你的驱动是从哪找的，我也在找驱动，一直没找到。

zmz125000, 2019-08-15:
https://github.com/openwrt/openwrt/commit/7d684b7673a7cb4f3fef366dd4f621f62fc4019c

GHSY123, 2019-08-15:
> [openwrt/openwrt@7d684b7](https://github.com/openwrt/openwrt/commit/7d684b7673a7cb4f3fef366dd4f621f62fc4019c)

你的2.4g和5g能同时工作吗？

zmz125000, 2019-08-15:
but not yet concurrently

GHSY123, 2019-08-15:
> but not yet concurrently

也是不能同时

lysssssss, 2019-08-17:
获取不到IP的问题解决了吗，我也遇到了，很烦恼。
不知道设置静态IP行不行呢

zmz125000, 2019-08-17:
不行

MeIsReallyBa, 2019-08-18:
双7615的机器能获取到ip，但效果非常差，超过路由器3m一测速wifi就断开。老老实实闭源驱动，开源驱动今年别想稳定了，维护的人太少

lysssssss, 2019-08-18:
> 双7615的机器能获取到ip，但效果非常差，超过路由器3m一测速wifi就断开。老老实实闭源驱动，开源驱动今年别想稳定了，维护的人太少

大佬有闭源驱动的教程吗，我按MTK那个教程编译出来每6小时卡死一次，而且编译完成后不会出现sysupgrade的固件

closed 2019-10-18 by coolsnowwolf