coolsnowwolf/lede #1547: HC5962如何开启128M

LondyGhost, 2019-08-15:
在 /target/linux/ramips/image中的 `7621.mk文件，204行中默认为32M，但是HC5962这个设备是128M的NAND，直接将32M改成128M只是在编译时候大小超过32M不会报警。但是刷入路由器后，空间不变
`IMAGE_SIZE := $(ralink_default_fw_size_32M)`

mk文件附上HC5962的段落。
`define Device/hc5962
  DTS := HC5962
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE := 2097152
  UBINIZE_OPTS := -E 5
  IMAGE_SIZE := $(ralink_default_fw_size_32M)
  IMAGES += factory.bin
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  IMAGE/factory.bin := append-kernel | pad-to $$(KERNEL_SIZE) | append-ubi | check-size $$$$(IMAGE_SIZE)
  DEVICE_TITLE := HiWiFi HC5962
  DEVICE_PACKAGES := kmod-mt7603 kmod-mt76x2 kmod-usb3 wpad-basic
endef
TARGET_DEVICES += hc5962`

programmejava, 2019-08-17:
修改你源码的/target/linux/ramips/dts/hc5962.dts文件，分区代码替换成去下：
                partition@0 {
                        label = "u-boot";
                        reg = <0x0 0x80000>;
                };

                partition@80000 {
                        label = "debug";
                        reg = <0x80000 0x80000>;
                };

                factory: partition@100000 {
                        label = "factory";
                        reg = <0x100000 0x40000>;
                };

                partition@140000 {
                        label = "kernel";
                        reg = <0x140000 0x200000>;
                };

                partition@340000 {
                        label = "ubi";
                        reg = <0x340000 0x7c40000>;
                };
编译后的固件可以剩余100M左右的空间。

调整最后label = "ubi"下面那一行

LondyGhost, 2019-08-17:
> 修改你源码的/target/linux/ramips/dts/hc5962.dts文件，分区代码替换成去下：
> partition@0 {
> label = "u-boot";
> reg = <0x0 0x80000>;
> };
>
> ```
>             partition@80000 {
>                     label = [...]

那么7621.mk这个文件需要修改对应的值吗？比如我写的32M改成128M这里还需要更改吗？这里只涉及包生成大小会不会超出范围而已吗？

programmejava, 2019-08-18:
只改dts文件就可以

alphavip, 2019-09-04:
> 只改dts文件就可以

ubi下边还有其他分区啊

programmejava, 2019-09-05:
只改partition@340000 {
                    label = "ubi";
                    reg = <0x340000 0x7c40000>;这里就可以，基它不用管，我改好已经一个多月了，正常使用  极4在没有用它全部分区时，固件会把剩余设置为/tmp分区，

closed 2019-10-18 by coolsnowwolf