coolsnowwolf/lede #1550: HC5962 USB2/USB3可以挂载3T的NTFS硬盘但是无法正常 samba

LondyGhost, 2019-08-18:
系统日志
```
Wed Aug 14 18:46:40 2019 daemon.err uhttpd[2886]: luci: accepted login on / for root from 192.168.2.192
Wed Aug 14 18:47:25 2019 kern.info kernel: [  183.745548] usb 2-1: new SuperSpeed USB device number 2 using xhci-mtk
Wed Aug 14 18:47:25 2019 kern.info kernel: [  183.781995] usb-storage 2-1:1.0: USB Mass Storage device detected
Wed Aug 14 18:47:25 2019 kern.info kernel: [  183.789174] scsi host0: usb-storage 2-1:1.0
Wed Aug 14 18:47:27 2019 kern.info kernel: [  185.580047] usb 1-1: new high-speed USB device number 2 using xhci-mtk
Wed Aug 14 18:47:27 2019 kern.info kernel: [  185.940065] usb 2-1: USB disconnect, device number 2
Wed Aug 14 18:47:36 2019 kern.err kernel: [  195.140082] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Wed Aug 14 18:47:36 2019 kern.info kernel: [  195.290112] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
Wed Aug 14 18:47:37 2019 kern.info kernel: [  195.321975] usb-storage 2-1:1.0: USB Mass Storage device detected
Wed Aug 14 18:47:37 2019 kern.info kernel: [  195.329050] scsi host0: usb-storage 2-1:1.0
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.405437] scsi 0:0:0:0: Direct-Access     ACASIS   BA-09USE         MX6O PQ: 0 ANSI: 0
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.415803] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.423813] sd 0:0:0:0: [sda] 5860533168 512-byte logical blocks: (3.00 TB/2.73 TiB)
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.432105] sd 0:0:0:0: [sda] Write Protect is off
Wed Aug 14 18:47:39 2019 kern.debug kernel: [  197.436883] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
Wed Aug 14 18:47:39 2019 kern.err kernel: [  197.437223] sd 0:0:0:0: [sda] No Caching mode page found
Wed Aug 14 18:47:39 2019 kern.err kernel: [  197.442568] sd 0:0:0:0: [sda] Assuming drive cache: write through
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.449694] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
Wed Aug 14 18:47:39 2019 kern.info kernel: [  197.511972]  sda: sda1 sda2
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.518509] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
Wed Aug 14 18:47:39 2019 kern.notice kernel: [  197.527835] sd 0:0:0:0: [sda] Attached SCSI disk
Wed Aug 14 18:47:41 2019 kern.err kernel: [  199.660091] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Wed Aug 14 18:47:41 2019 kern.info kernel: [  199.666845] usb usb1-port1: attempt power cycle
Wed Aug 14 18:47:46 2019 kern.err kernel: [  204.350100] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Wed Aug 14 18:47:50 2019 kern.err kernel: [  208.690077] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
Wed Aug 14 18:47:50 2019 kern.err kernel: [  208.696866] usb usb1-port1: unable to enumerate USB device
Wed Aug 14 18:48:10 2019 kern.warn kernel: [  228.697152] [EXFAT] trying to mount...
Wed Aug 14 18:48:10 2019 kern.err kernel: [  228.709682] ntfs: (device sda1): parse_options(): Unrecognized mount option discard.
Wed Aug 14 18:48:40 2019 kern.err kernel: [  259.106184] ntfs: (device sda2): parse_ntfs_boot_sector(): Volume size (2TiB) is too large for this architecture.  Maximum supported is 2TiB.  Sorry.
Wed Aug 14 18:48:40 2019 kern.err kernel: [  259.119704] ntfs: (device sda2): ntfs_fill_super(): Unsupported NTFS filesystem.
Wed Aug 14 18:48:40 2019 daemon.err block: mounting /dev/sda2 (ntfs) as /mnt/sda2 failed (22) - Invalid argument

```

内核日志

```
[    6.758420] mtk_soc_eth 1e100000.ethernet eth0: port 1 link up
[    6.906458] kmodloader: loading kernel modules from /etc/modules-boot.d/*
[    7.054681] usbcore: registered new interface driver usbfs
[    7.060309] usbcore: registered new interface driver hub
[    7.065777] usbcore: registered new device driver usb
[    7.074518] exFAT: Version 1.2.9
[    7.102378] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    7.115370] SCSI subsystem initialized
[    7.122727] ehci-platform: EHCI generic platform driver
[    7.129137] ehci-pci: EHCI PCI platform driver
[    7.135512] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    7.142759] ohci-platform: OHCI generic platform driver
[    7.149671] uhci_hcd: USB Universal Host Controller Interface driver
[    7.157231] ohci-pci: OHCI PCI platform driver
[    7.169883] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vbus not found, using dummy regulator
[    7.178632] xhci-mtk 1e1c0000.xhci: 1e1c0000.xhci supply vusb33 not found, using dummy regulator
[    7.187570] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    7.192951] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
[    7.210144] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000210010
[    7.219311] xhci-mtk 1e1c0000.xhci: irq 19, io mem 0x1e1c0000
[    7.226022] hub 1-0:1.0: USB hub found
[    7.229839] hub 1-0:1.0: 2 ports detected
[    7.234486] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    7.239797] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
[    7.247275] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0  SuperSpeed
[    7.253935] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    7.262948] hub 2-0:1.0: USB hub found
[    7.266768] hub 2-0:1.0: 1 port detected
[    7.276047] usbcore: registered new interface driver usb-storage
[    7.282434] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
[    7.300295] init: - preinit -
[... 126 lines cut ...]
[  195.140082] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  195.290112] usb 2-1: new SuperSpeed USB device number 3 using xhci-mtk
[  195.321975] usb-storage 2-1:1.0: USB Mass Storage device detected
[  195.329050] scsi host0: usb-storage 2-1:1.0
[  197.405437] scsi 0:0:0:0: Direct-Access     ACASIS   BA-09USE         MX6O PQ: 0 ANSI: 0
[  197.415803] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  197.423813] sd 0:0:0:0: [sda] 5860533168 512-byte logical blocks: (3.00 TB/2.73 TiB)
[  197.432105] sd 0:0:0:0: [sda] Write Protect is off
[  197.436883] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[  197.437223] sd 0:0:0:0: [sda] No Caching mode page found
[  197.442568] sd 0:0:0:0: [sda] Assuming drive cache: write through
[  197.449694] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  197.511972]  sda: sda1 sda2
[  197.518509] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  197.527835] sd 0:0:0:0: [sda] Attached SCSI disk
[  199.660091] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  199.666845] usb usb1-port1: attempt power cycle
[  204.350100] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  208.690077] usb usb1-port1: Cannot enable. Maybe the USB cable is bad?
[  208.696866] usb usb1-port1: unable to enumerate USB device
[  228.697152] [EXFAT] trying to mount...
[  228.709682] ntfs: (device sda1): parse_options(): Unrecognized mount option discard.
**[  259.106184] ntfs: (device sda2): parse_ntfs_boot_sector(): Volume size (2TiB) is too large for this architecture.  Maximum supported is 2TiB.  Sorry.**
[  259.119704] ntfs: (device sda2): ntfs_fill_super(): Unsupported NTFS filesystem.
```

nivalxer, 2019-08-18:
内核日志最后几行有写到：Volume size (2TiB) is too large for this architecture.  Maximum supported is 2TiB.  Sorry.

LondyGhost, 2019-08-18:
> 内核日志最后几行有写到：Volume size (2TiB) is too large for this architecture. Maximum supported is 2TiB. Sorry.

我知道。但是不知道是什么原因导致的。官方openwrt是可以挂载成功的。

closed 2019-10-18 by coolsnowwolf