coolsnowwolf/lede #1552: English Please

hobolance, 2019-08-19:
### What is the problem with Chinese README's

Firstly, we congratulate you for getting so much star by sharing this repository with humanity.

But it is very disappointing for non-Chinese speakers when one couldn't understand what a trending repository is about.

When we see such a repo on trending, our minds are blurring like Gollum's.

[image]

There is a way **you can help** to solve this disappointment which I believe is experienced by many people who want to know more about your valuable work and appreciate it.

### What we want:
 - Please add English translation of your README so you are sharing your work and knowledge with more people.

### How this will help you:
 - More feedback to fix and improve your project.
 - New ideas about your project.
 - Greater fame.
 - [image]

---

_“Sharing knowledge is the most fundamental act of friendship. Because it is a way you can give something without loosing something.”_

_**— Richard Stallman**_

**Thank you!**	

> The developers of repo will not be held responsible for problems that arise in this script.

> This issue generated by [us/english-please](github.com/us/english-please) script. Please open issue on any error. Thank you!

pys078, 2019-08-19:
As a chinese,i support you.

KFERMercer, 2019-08-19:
In fact, there is an English translation of README below.

KFERMercer, 2019-08-19:
Please don't send this script everywhere

closed 2019-08-19 by hobolance