coolsnowwolf/lede #1559: u_env/s_env有问题，报Bad CRC

dunkeng, 2019-08-21:
路由器：WRT1200AC，代码最新
问题：刷固件后，u_env或者是s_env有问题，导致一无法刷固件，无法切换分区等问题
root@OpenWrt:~# fw_printenv 
Cannot parse config file '/etc/fw_env.config': No such file or directory
补文件后：
root@OpenWrt:/tmp# fw_printenv boot_part
Warning: Bad CRC, using default environment
## Error: "boot_part" not defined
root@OpenWrt:/tmp# fw_printenv 
Warning: Bad CRC, using default environment
bootcmd=bootp; setenv bootargs root=/dev/nfs nfsroot=${serverip}:${rootpath} ip=${ipaddr}:${serverip}:${gatewayip}:${netmask}:${hostname}::off; bootm
bootdelay=5
baudrate=115200
root@OpenWrt:/tmp# fw_setenv boot_part 2
Warning: Bad CRC, using default environment
MTD erase error on /dev/mtd2: Invalid argument
Error: can't write fw_env to flash
root@OpenWrt:/tmp# fw_setenv boot_part 1
Warning: Bad CRC, using default environment
MTD erase error on /dev/mtd2: Invalid argument
Error: can't write fw_env to flash

附件是1900的正常ENV，不确认和1200是不是完全一样的，导入可用
[1900ac_uboot_env.txt](https://github.com/coolsnowwolf/lede/files/3524987/1900ac_uboot_env.txt)

[wrt1200ac.zip](https://github.com/coolsnowwolf/lede/files/3524990/wrt1200ac.zip)

dunkeng, 2019-08-21:
OpenWrt R9.8.17 / LuCI Master (git-19.146.62144-fd6fdb2)

dunkeng, 2019-08-21:
回滚9eeb898ee39567d5eef5d7e87c3db062ce9c860b版本，一切正常

dunkeng, 2019-09-06:
新版本问题解决

closed 2019-09-06 by dunkeng

yebr, 2019-09-15:
我在OpenWrt R9.8.17 / LuCI Master (git-19.146.62144-fd6fdb2)遇到同样的bug，请问怎样导入u_env的s_env？