coolsnowwolf/lede #1562: luci-app-ssr-plus uci: Entry not found

pcccccc, 2019-08-21:
安装luci-app-ssr-plus 时报错：uci: Entry not found。
日志`root@Openwrt:/# opkg install luci-app-ssr-plus_1-98_all.ipk                                                                 
Installing luci-app-ssr-plus (1-98) to root...
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Installing shadowsocks-libev-config (3.3.1-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/shadowsocks-libev-config_3.3.1-1_x86_64.ipk
Installing shadowsocks-libev-ss-redir (3.3.1-1) to root...
Downloading http://downloads.openwrt.org/snapshots/packages/x86_64/packages/shadowsocks-libev-ss-redir_3.3.1-1_x86_64.ipk
Configuring shadowsocks-libev-config.
Configuring shadowsocks-libev-ss-redir.
Configuring luci-app-ssr-plus.
uci: Entry not found`
平台X64

coolsnowwolf, 2019-08-22:
不是报错

closed 2019-08-22 by coolsnowwolf

pcccccc, 2019-08-22:
那在菜单里咋看不到选项呢。。

pcccccc, 2019-08-22:
> 不是报错

那在菜单里咋看不到选项呢。。

xman55555, 2019-08-24:
[image]
这种包找不到要怎么手动下载呢?

richcup, 2019-10-17:
luci-app-ssr-plus安装报错
uci: Entry not found
uci: Entry not found
uci: Entry not found
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/dhcp is different from the conffile in the new package. The new conffile will be placed at /etc/config/dhcp-opkg.
服務端頁面 ( 無法顯示) 其它都正常
[image]
[image]

Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocksr/server'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/shadowsocksr/server.lua:7: module 'luci.model.ipkg' not found:
如何修復

gause555, 2020-03-14:
先运行opkg update正常就可以解决继续安装

sobossceo, 2020-05-20:
Installing luci-app-ssr-plus (1-100) to root...
Configuring luci-app-ssr-plus.
uci: Entry not found
uci: Parse error (invalid command) at line 1, byte 0

70599, 2020-12-13:
`uci: Entry not found`是因为luci-app-ssr-plus把非uci配置文件，例如`black.list`之类的放入了`/etc/config`路径下导致的。
这个警告信息本身不影响插件运行。
如果不能运行请排查其他因素。