coolsnowwolf/lede #1568: pppoe获取ipv6后不会自动生成v6接口

colinhd8, 2019-08-23:
官方文档([https://openwrt.org/docs/guide-user/network/ipv6/start])有一段这样的内容：
[image]
，而其它固件实际使用下来也确实是这样的：
[image]
但大雕的固件中(本周一获取最新源码编译的)是这样的(没有_6的接口)
[image]
当然，在概览页是可以看到V6信息的：
[image]
这是因为luci版本旧的原因吗？有啥方法可以保持和官方的一致呢？

rufengsuixing, 2019-09-17:
不用生成也能用，一个接口能有v4和v6同时存在，关了吧

colinhd8, 2019-09-19:
已确认其实是有生成了_6接口，只是没在页面上显示，应该是luci比较旧的原因。

closed 2019-09-19 by colinhd8