coolsnowwolf/lede #1572: 请教，raspi3ssr-plus complie 报错

crackself, 2019-08-25:
自构建的LFS系统，target是raspi3 B+，主机gcc8.4，构建的交叉工具链是gcc8.3
使用arm32版本（raspi2b~3b+适用）编译是遇到一下错误：
在使用aarch64版本编译无此错误。
```
/home/lfs/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/libudns'
Making all in libev
make[5]: Entering directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/libev'
  CC       ev.lo
ev.c:1825:31: warning: 'ev_default_loop_ptr' initialized and declared 'extern'
   EV_API_DECL struct ev_loop *ev_default_loop_ptr = 0; /* needs to be initialised to make it a definition despite extern */
                               ^~~~~~~~~~~~~~~~~~~
  CC       event.lo
  CCLD     libev.la
OpenWrt-libtool: link: warning: `-version-info/-version-number' is ignored for convenience libraries
/home/lfs/openwrt/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-8.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/8.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/libev'
Making all in src
make[5]: Entering directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
  CC       libshadowsocks_libev_la-utils.lo
  CC       libshadowsocks_libev_la-jconf.lo
  CC       libshadowsocks_libev_la-json.lo
  CC       libshadowsocks_libev_la-encrypt.lo
  CC       libshadowsocks_libev_la-udprelay.lo
  CC       libshadowsocks_libev_la-cache.lo
  CC       libshadowsocks_libev_la-acl.lo
  CC       libshadowsocks_libev_la-netutils.lo
  CC       libshadowsocks_libev_la-local.lo
local.c: In function 'server_recv_cb':
local.c:721:45: error: argument to 'sizeof' in 'strncpy' call is the same expression as the source; did you mean to use the size of the destination? [-Werror=sizeof-pointer-memaccess]
                     strncpy(host, ip, sizeof(ip));
                                             ^
cc1: all warnings being treated as errors
make[5]: *** [Makefile:791: libshadowsocks_libev_la-local.lo] Error 1
make[5]: Leaving directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/src'
make[4]: *** [Makefile:479: all-recursive] Error 1
make[4]: Leaving directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
make[3]: *** [Makefile:388: all] Error 2
make[3]: Leaving directory '/home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev/shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543'
make[2]: *** [Makefile:67: /home/lfs/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/shadowsocksr-libev//shadowsocksr-libev-2.5.6-d63ff863800a5645aca4309d5dd5962bd1e95543/.built] Error 2
make[2]: Leaving directory '/home/lfs/openwrt/package/crackself/shadowsocksr-libev'
time: package/crackself/shadowsocksr-libev/compile#78.99#27.83#100.59
make[1]: *** [package/Makefile:113: package/crackself/shadowsocksr-libev/compile] Error 2
make[1]: Leaving directory '/home/lfs/openwrt'
make: *** [/home/lfs/openwrt/include/toplevel.mk:218: package/luci-app-ssr-plus/compile] Error 2
```

closed 2019-08-31 by crackself

wuxiaowei, 2019-10-12:
你好这个问题 你解决了吗

crackself, 2019-10-12:
v> 你好这个问题 你解决了吗

没，我已放弃。改用openwrt官方源、v2ray。