coolsnowwolf/lede #1581: luci-app-ssr-pro 文件缺失

275761919, 2019-08-29:
我没有用完整的环境编译。用openwrt-sdk官方环境编译的，应该没什么影响吧，只是不想重新烧固件
打开 shadowsocksR 时提示：
```
/usr/lib/lua/luci/cbi.lua:53: Model 'shadowsocksr/general' not found!
```
以及
```
/usr/lib/lua/luci/cbi.lua:53: Model 'shadowsocksr/access-control' not found!
```

275761919, 2019-08-30:
已经执行命令 `echo 0xDEADBEEF > /etc/config/google_fu_mode`

275761919, 2019-08-30:
执行命令 `echo 0xDEADBEEF > /etc/config/google_fu_mode`后重新安装提示：
```
uci: Entry not found
uci: Parse error (invalid command) at line 1, byte 0
```
使用https://github.com/coolsnowwolf/lede/issues/1562 中的`shadowsocks.lua`同样提示
```
/usr/lib/lua/luci/cbi.lua:53: Model 'shadowsocksr/general' not found!
```

ope087, 2019-09-01:
这两个文件不是这个组件里的,也不在这个位置,general应该是dnsforwarder的;虽然有accesscontrol的组件,但是access-control好像不是里面的.

275761919, 2019-09-02:
@ope087 那会是luci版本不对之类的吗，老提示`/usr/lib/lua/luci/cbi.lua:53: Model 'shadowsocksr/general' not found!`和`uci: Parse error (invalid command) at line 1, byte 0`

ope087, 2019-09-02:
你把luci-app-dnsforwarder一起编译进去

ope087, 2019-09-02:
pdnsd-alt也加上吧,有些组件没在makefile里做依赖

275761919, 2019-09-02:
我有点想明白了，好像是以前装的另一版本app的遗留问题，我重新编译固件试试吧，非常感谢 @ope087

275761919, 2019-09-02:
确定了，前装的另一版本app的遗留问题

closed 2019-09-03 by 275761919