coolsnowwolf/lede #1582: ss中缺少对obfs和v2ray两个插件的支持，仅仅剩下kcptun

moralrebuild, 2019-08-29:
这两个插件目前还是比较抗封锁的，相对于原生v2ray配置简单门槛低，适合草根用户。请考虑提供支持。非常感谢提供精良高质量的软件帮助大陆网民！

275761919, 2019-08-30:
可以发我一份 shadowsocks 的 general.lua 和 access-control.lua 给我吗，我编译出来的少了这两个文件，在 /usr/lib/lua/luci/model/cbi/shadowsocksr 路径下 十分感谢 邮箱 liao9256@163.com

moralrebuild, 2019-08-30:
抱歉我没有编译L大的固件，我通常只编译官方的。L大的固件功能很好，可惜因不可描述原因，不能并入主流openwrt，这也是我对部分国人不能充分理解真正开源精神的悲凉感。自己的明明踩在别人的肩膀上却总要求别人别踩他

275761919, 2019-08-30:
我也没有编译L大的固件，我用lede-sdk编译了luci-app-ssr-plus的包，安装后缺少上述文件，please！

275761919, 2019-08-30:
比较好奇已经开源，还不能并入openwrt的主流的原因

moralrebuild, 2019-08-30:
我觉得应该不是能力不够。剩下的原因无非就是自信心或者私心了

shikkoku, 2019-08-30:
v2ray Plugs有哪家机场可以支持？

moralrebuild, 2019-08-30:
我也不用公共机场，自己搭建私人的。来这儿提一些建议是为了增加开源项目的透明度和功能。如果这些都有改善，我有朝一日也会用的。

zxlhhyccc, 2019-09-01:
> 这两个插件目前还是比较抗封锁的，相对于原生v2ray配置简单门槛低，适合草根用户。请考虑提供支持。非常感谢提供精良高质量的软件帮助大陆网民！

已经有这两个源码的SSR+

moralrebuild, 2019-09-01:
@zxlhhyccc 请告诉具体是哪个版本或者在哪个另外的开发者链接下？

coolsnowwolf, 2019-09-01:
其实我们用的 OpenWrt 功能需求和其他国家差异过大是主要的原因，这里不细说了，你应该也懂的。

是否合并到主干并不是技术问题。

有一处对说中文的人开源的 OpenWrt Fork其实也不错吧？

coolsnowwolf, 2019-09-01:
另外说一下，我本人对于追新功能，除非经过了充分的测试和论证才会考虑支持

moralrebuild, 2019-09-02:
@coolsnowwolf 伊朗、叙利亚等美国眼里的流氓国家也在用中国发明的GFW技术，将来可能北朝鲜人民能使用互联网的话，GFW也是题中应有之义。所以大牛可以考虑兼顾国际友人的需求。说实话，我觉得仅仅从安全漏洞角度考虑去维护一个单独的openwrt是吃力不讨好的，何开新功能的跟进。我担心有一天你还需要花费难以预知的精力移植到19.07,20.x.., 25.y等等主流版本上，或者干脆放弃。那就太可惜了你的辛苦劳动。个人意见，可能有偏颇。

moralrebuild, 2019-09-02:
SS/SSR/V2RAY作为通用的抗封锁工具，被中国人的聪明才智发扬光大。相信有一天西方人主流用户会意识到有一种能和通用VPN分庭抗礼的安全上网技术。加油吧，我觉得这是为数不多的的来自中国人对IT技术的贡献。

KFERMercer, 2019-09-02:
建议尽量不要自我感动:expressionless:

tufu9441, 2019-09-20:
> 这两个插件目前还是比较抗封锁的，相对于原生v2ray配置简单门槛低
@moralrebuild 的看法我也赞同，现在ss+obfs还是比较稳的，反而是机场广泛使用的ssr经常大面积被封。

tufu9441, 2019-09-24:
@moralrebuild 不过想用obfs的话，用Lean大的源码编译也可以实现，直接把 @aa65535 做的[SS](https://github.com/shadowsocks/openwrt-shadowsocks)和[simple-obfs](https://github.com/aa65535/openwrt-simple-obfs)放在packages下面，编译出来就能用。同时根据需求还可以搭配[ChinaDNS](https://github.com/aa65535/openwrt-chinadns)和[dns-forwarder](https://github.com/aa65535/openwrt-dns-forwarder)，并且都是有luci界面的。

closed 2019-09-29 by coolsnowwolf