coolsnowwolf/lede #1586: BitComet下载时负载高，K2P路由器卡死

xynu617, 2019-08-31:
8月30日晚上用最新代码编译的，插件只选了SSR-PLUS，ADBYBY-PLUS KMS服务器 Turbo ACC UPNP，去掉了无线驱动，只要电脑一开BitComet，路由器概况界面的活动连接数和实时负载就飙升，连接数最高到5000多，负载到7左右，然后就无法访问路由器了，关掉BitComet过一会路由器就会恢复
[image]
[image]

ope087, 2019-09-01:
SSR-PLUS，ADBYBY-PLUS , Turbo ACC ,UPNP一个一个关了试试

wuypinfo, 2019-09-01:
换N1来下载。路由都被累死了。

coolsnowwolf, 2019-09-01:
CPU 拉不住这么多 UDP Fullcone 了，升级 CPU 或者关闭 Fullcone 。建议升级 CPU

xynu617, 2019-09-04:
> CPU 拉不住这么多 UDP Fullcone 了，升级 CPU 或者关闭 Fullcone 。建议升级 CPU

多谢大雕，关了FullCone-NAT果然就好了

closed 2019-09-04 by xynu617